ros-visualization/rviz #1622: MotionPlanning display prevents RViz from opening when no message are received on /clock

christian-rauch, 2021-04-15:
When having `/use_sim_time` set and no bag file is played, the MotionPlanning display prevents RVi from showing the GUI.

Reproduce:
1. prepare a `my_robot.rviz` file with the `MotionPlanning` display added, it can be deactivated by needs to be in the list
2. `rosparam set /use_sim_time true`
3. `rviz -d my_robot.rviz`

The small splash screen will be shown, but the GUI with the 3D view will only open after a bag file is played.

### Your environment
* OS Version: e.g. Ubuntu 20.04
* ROS Distro: Noetic
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
```
rviz version 1.14.5
compiled against Qt version 5.12.8
compiled against OGRE version 1.9.0 (Ghadamon)
Forcing OpenGl version 0.
Stereo is NOT SUPPORTED
OpenGL device: Mesa Intel(R) UHD Graphics 630 (CFL GT2)
OpenGl version: 4.6 (GLSL 4.6) limited to GLSL 1.4 on Mesa system.
```

rhaschke, 2021-04-15:
The `MotionPlanningDisplay` attempts to create several `ActionClients`. Doing so [requires a valid time signal](https://github.com/ros/actionlib/blob/e1338aa4c74f79b320e88952df9e1df5c432fce6/actionlib/include/actionlib/client/action_client.h) as introduced in https://github.com/ros/actionlib/pull/62. I don't see how this could be solved other than correctly providing the time signal. In any case, this is an issue related to MoveIt. How would you expect rviz to handle stuck plugins?

christian-rauch, 2021-04-16:
Ok, I did not realise that this all runs in a single thread. So if any plugin decides to wait for something, the entire GUI will freeze? In such a case, this is then an architectural problem since I think that these plugins should not be able to make the GUI unresponsive to user input.

rhaschke, 2021-04-16:
In principle, I agree. On the other hand, all GUI stuff needs to run in the main GUI thread. So, there is no easy way out from this dilemma.

closed 2021-04-16 by rhaschke