ros-visualization/rviz #1625: Point Cloud shown with ghost points when transformed

christian-rauch, 2021-04-28:
Publishing an unordered point cloud (`sensor_msgs/PointCloud2`) in RViz leaves trails (or ghost points):
[image]

This screenshot is showing the point cloud in RViz with trails or "ghost" points, next to the rendered view of how the cloud is supposed to look like from the camera view, and with the point cloud as visualised by https://github.com/dirkholz/pcl_online_viewer.

My queue size and decay time are 1. This should mean that only the newest point cloud message is ever considered.

After disabling and enabling the display type, the point cloud is correctly shown for the next frame, but it then accumulates points again.

The point cloud is defined in the camera frame, while RViz projects everything into the world frame. When setting the camera frame directly as the scene root (i.e. no transformations required) the point cloud is also shown without ghost points. So this seems to be directly related to the transformations.

### Your environment
Ubuntu 20.04, noetic
```
rviz version 1.14.5
compiled against Qt version 5.12.8
compiled against OGRE version 1.9.0 (Ghadamon)
Forcing OpenGl version 0.
```

rhaschke, 2021-06-17:
Do I understand correctly, that the object is moving and "old" points are still shown?

christian-rauch, 2021-06-17:
> Do I understand correctly, that the object is moving and "old" points are still shown?

Yes, that's correct. In reality, it is only a single points clouds at one point in time. But the transformations somehow lead to this overlap of points clouds from multiple points in time.

rhaschke, 2021-06-19:
I cannot confirm this issue. Using a [simple point cloud publisher](https://gist.github.com/rhaschke/64004339c49dd13f84f6dd1efa6e532a), I wasn't able to reproduce the issue.

> My queue size and decay time are 1. This should mean that only the newest point cloud message is ever considered.

I think that's the culprit: You need to use a decay time of zero!
Also, a ROS message queue size of 2 is recommended to not risk losing messages.

closed 2021-06-19 by rhaschke

christian-rauch, 2021-06-24:
@rhaschke Thank you for your efforts in trying to solve this issue.

I am uploading a 10s snippet of my log file ([pc_tf.zip](https://github.com/ros-visualization/rviz/files/6708429/pc_tf.zip)):
```
topics:      /point_cloud     64 msgs    : sensor_msgs/PointCloud2
             /tf            4404 msgs    : tf2_msgs/TFMessage     
             /tf_static        3 msgs    : tf2_msgs/TFMessage
```

The topic `/point_cloud` contains a moving point cloud in the `rgb_camera_link` frame.

If you replay this with `rosbag play --clock pc_tf.bag` (and `rosparam set /use_sim_time true`) you should see the duplicated point clouds in RViz when using the `world` as the fixed frame. The transformation from `/world` -> `/rgb_camera_link` is constant during this sniped, so time synchronisation should also not be an issue.

rhaschke, 2021-06-24:
Sorry, but I still cannot see any duplicated points. However, I noticed that the number of points in the point cloud varies between 1300 and 2400 points! Might that be the reason?
Please consider my simple example from https://github.com/ros-visualization/rviz/issues/1625#issuecomment-864347099.
From your rosbag it is not possible to know how many points should be visible.

christian-rauch, 2021-06-24:
Yes, the point cloud size changes over time. However, if you just replay the bagfile with the `world` (RViz has to transform the points) and `rgb_camera_link` (no transformation needed) as the fixed frame, and compare the shown points, you can clearly see that there are duplicated points when the `world` is used as the fixed frame and RViz has to transform the points. It's not about the number of points in the individual point cloud messages, it's about the "shown" points after transformation in RViz.

rhaschke, 2021-06-24:
Just to clarify what I see: Here is a video of both configs: left - `rgb_camera_link`, right - `world`. Apart from a slight delay (probably due to extra scheduling for resolving the transformation), I don't really see a difference.

https://user-images.githubusercontent.com/5376030/123259164-77af0680-d4f4-11eb-807d-4b38db972d8c.mp4

christian-rauch, 2021-06-24:
I just learned that the `Decay Time` is in seconds and not number of messages, and that only 0 means to use the most recent message. I mentioned that in the original issue:
> My queue size and decay time are 1. This should mean that only the newest point cloud message is ever considered.

and thought that this setting is correct.

But obviously, it is not. After setting the decay time to 0, everything looks as expected.

I am so sorry to take your time, now that I see that it was a configuration issue on my side.

rhaschke, 2021-06-24:
That's what I pointed out in https://github.com/ros-visualization/rviz/issues/1625#issuecomment-864347099 already :wink:
No worries. Nice that we finally figured out the issue.

christian-rauch, 2021-06-24:
> That's what I pointed out in [#1625 (comment)](https://github.com/ros-visualization/rviz/issues/1625#issuecomment-864347099) already

Oops. I looked at the script and forgot to read the rest of the message :-)