ros-visualization/rviz #1626: PointCloud2 Display: Cannot manually set min/max intensity

johuber, 2021-04-30:
After upgrading from ROS melodic to noetic I cannot set the min/max intensity values anymore in a PointCloud2 display, when I untick "Autocompute Intensity Bounds".  (see highlighted field in rviz screenshot).

- Even manually editing the min/max values in the `.rviz` file does not work, when I change the bounds there they will be loaded as [0, 4096].
- when I tick and untick "Autocompute Intensity Bounds" (min/max values will update to some values) and then save the rviz config, the bounds in the rviz file don't change their value.

How to reproduce:
1. subscribe to any pointcloud2 topic
2. set "Color Transformer" to "intensity"
3. Untick "Autocompute Intensity bounds"
4. try to set min or max intensity value  --> does not work
5. save your rviz config and see if the min/max values corresponds to what it currently says in RVIZ
6. change the values in the .rviz file and reload the config --> see min/max values change to [0,4096]

Describe your issue here and explain how to reproduce it.

### Your environment
* OS Version: e.g. Ubuntu 20.04
* ROS Distro: [Noetic]
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
    ```
    [ INFO] [1619766592.458320745][${node}][/tmp/binarydeb/ros-noetic-rviz-1.14.5/src/rviz/visualizer_app.cpp:114]: rviz version 1.14.5
[ INFO] [1619766592.458375693][${node}][/tmp/binarydeb/ros-noetic-rviz-1.14.5/src/rviz/visualizer_app.cpp:115]: compiled against Qt version 5.12.8
[ INFO] [1619766592.458391012][${node}][/tmp/binarydeb/ros-noetic-rviz-1.14.5/src/rviz/visualizer_app.cpp:116]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1619766603.100651819][/rviz_1619766592458490660][/tmp/binarydeb/ros-noetic-rviz-1.14.5/src/rviz/ogre_helpers/render_system.cpp:84]: Forcing OpenGl version 0.
[ INFO] [1619766606.351965812][/rviz_1619766592458490660][/tmp/binarydeb/ros-noetic-rviz-1.14.5/src/rviz/ogre_helpers/render_system.cpp:466]: Stereo is NOT SUPPORTED
[ INFO] [1619766606.352046530][/rviz_1619766592458490660][/tmp/binarydeb/ros-noetic-rviz-1.14.5/src/rviz/ogre_helpers/render_system.cpp:181]: OpenGL device: GeForce GTX 1070/PCIe/SSE2
[ INFO] [1619766606.352102245][/rviz_1619766592458490660][/tmp/binarydeb/ros-noetic-rviz-1.14.5/src/rviz/ogre_helpers/render_system.cpp:227]: OpenGl version: 4,6 (GLSL 4,6).

    ```
[image]

rhaschke, 2021-04-30:
Should be fixed in release 1.14.7 which is not yet synced. Try the [ros-testing](http://wiki.ros.org/action/show/TestingRepository) repos?

johuber, 2021-04-30:
> Should be fixed in release 1.14.7 which is not yet synced. Try the [ros-testing](http://wiki.ros.org/action/show/TestingRepository) repos?

oh nice, I will try it out, thank you!

johuber, 2021-04-30:
that fixed it! thanks a lot @rhaschke .

closed 2021-04-30 by johuber