ros-visualization/rviz #1628: In docker with ros noetic, rviz show black block when dragging the window.

JS00000, 2021-05-12:
When run rviz in docker container, anything about ros is ok, but the rviz render has issue. When I enlarge or reduce the window of rviz, some area in rviz will turn into black block. It seems that rviz do not refresh when the window is zoomed. This problem only happen in ros noetic. In ros melodic, everything is ok.

[image]

### Your environment
* Host OS Version: Both Ubuntu 18.04 and Ubuntu 20.04 are tested. The problem is the same.
* CPU: x86/amd64
* GPU: Both Nvidia 1060 and Integrated Graphics are tested.
* Docker Version: 20.10.6 build 370c289
* Docker image: Both osrf/ros:noetic-desktop-full and https://github.com/turlucode/ros-docker-gui/blob/master/nvidia/noetic/base/Dockerfile are tested.
* OS in docker: Ubuntu 20.04
* ROS Distro in docker: Noetic
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
    ```
➜  / rviz
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
[ INFO] [1620765053.829899259]: rviz version 1.14.7
[ INFO] [1620765053.829930451]: compiled against Qt version 5.12.8
[ INFO] [1620765053.829937498]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1620765053.836876324]: Forcing OpenGl version 0.
[ INFO] [1620765054.221535894]: Stereo is NOT SUPPORTED
[ INFO] [1620765054.221578974]: OpenGL device: GeForce GTX 1060/PCIe/SSE2
[ INFO] [1620765054.221594279]: OpenGl version: 4.6 (GLSL 4.6).
    ```

rhaschke, 2021-05-12:
To understand the issue, it would be helpful to know how you pass X11 permissions and GPU access to the container.
What is your host system?

JS00000, 2021-05-12:
My host system is ubuntu 18.04. It has an NVIDIA GPU 1060. And I install the nvidia-docker2 package in it.
I then run the docker container with GPU support in the following command:
```
imgName=osrf/ros:noetic-desktop-full
docker run -it -d --name ros_test \
					--privileged \
					--net=host \
					--gpus=all \
					--runtime=nvidia \
					-e NVIDIA_VISIBLE_DEVICES=all \
					-e NVIDIA_DRIVER_CAPABILITIES=all \
					-e DISPLAY \
					-e QT_X11_NO_MITSHM=1 \
					-v /tmp/.X11-unix:/tmp/.X11-unix:rw \
					-v /dev:/dev \
					$imgName
xhost +local:`docker inspect --format='{{ .Config.Hostname }}' ros_test`
docker exec -it ros_test bash
``` 
Another docker [image](https://github.com/turlucode/ros-docker-gui/blob/master/nvidia/noetic/base/Dockerfile) based on nvidia image is tested in this host, and the run script can be get in the [Readme](https://github.com/turlucode/ros-docker-gui/blob/master/Readme.md) of this rep. In this situation, the problem is the same.

In another host without NVIDIA GPU, whose system is ubuntu 20.04, I run the docker with:
```
imgName=osrf/ros:noetic-desktop-full
docker run -it -d --name ros_test \
					--privileged \
					--net=host \
					-e DISPLAY \
					-e QT_X11_NO_MITSHM=1 \
					-v /tmp/.X11-unix:/tmp/.X11-unix:rw \
					-v /dev:/dev \
					$imgName
xhost +local:`docker inspect --format='{{ .Config.Hostname }}' ros_test`
docker exec -it ros_test bash
```
The problem is the same.

JS00000, 2021-05-12:
> To understand the issue, it would be helpful to know how you pass X11 permissions and GPU access to the container.
> What is your host system?

I know the problem, maybe my run command is wrong. I have just used the docker run script in [this](https://github.com/turlucode/ros-docker-gui) without the problem.

JS00000, 2021-05-12:
> I know the problem, maybe my run command is wrong. I have just used the docker run script in [this](https://github.com/turlucode/ros-docker-gui) without the problem.

Actually I miss the argument ```--ipc=host```

rhaschke, 2021-05-12:
Is the problem resolved with adding the argument `--ipc=host`?
In any case, I'm afraid that this a Qt issue: All background issues occur in standard Qt widgets.

JS00000, 2021-05-12:
> Is the problem resolved with adding the argument `--ipc=host`?
> In any case, I'm afraid that this a Qt issue: All background issues occur in standard Qt widgets.

Yes, when I add the argument `--ipc=host` to the docker run command, the problem disappear. I didn't notice this argument because the docker image with ros melodic didn't need it.

rhaschke, 2021-05-12:
Good to know. Probably rendering employs some shared memory between client and X11 server now?

closed 2021-05-12 by rhaschke