ros-visualization/rviz #1633: rviz2 randomly crashes while running with nav2 stack

AlexeyMerzlyakov, 2021-06-08:
### Basic info
RViz2 from frequently crashes during the use while running with Nav2 stack. The problem appears in random places even if all Nav2-related stuff was disabled (see minimal configuration file in reproduction section below). Crashes were [noted](https://github.com/ros-planning/navigation2/issues/2367) on Crystal and still appearing on latest Rolling.

### More info
Crashes appears irregularly in different places. All types of messages and backtraces were detected and sorted-out by their probability of occurrence:
1. Invalid Argument
```
terminate called after throwing an instance of 'std::system_error'
  what():  Invalid argument
```
2. Segfault
```
Thread 16 "rviz2" received signal SIGSEGV, Segmentation fault.
```
3. Assertion Robust
```
rviz2: ../nptl/pthread_mutex_lock.c:428: __pthread_mutex_lock_full: Assertion `e != ESRCH || !robust' failed.
--Type <RET> for more, q to quit, c to continue without paging--
```
4. Deadlock: RViz2 hangs during the use
5. Assertion Prio
```
[rviz2-4] rviz2: tpp.c:82: __pthread_tpp_change_priority: Assertion `new_prio == -1 || (new_prio >= fifo_min_prio && new_prio <= fifo_max_prio)' failed.
[ERROR] [rviz2-4]: process has died [pid 174985, exit code -6, cmd '/home/leha/ros2_rolling/install/rviz2/lib/rviz2/rviz2 -d /home/leha/nav2_ws/install/nav2_bringup/share/nav2_bringup/rviz/nav2_default_view.rviz --ros-args'].
```

For the types No.1-4 GDB backtraces were collected and attached in the archive: [backtraces.zip](https://github.com/ros-visualization/rviz/files/6617895/backtraces.zip)
For type No.5 backtrace was not obtained (it was detected once). All of the problems above were detected by using the same scenario. Thus maybe (but not proven) all issue types have the same root cause.

### Steps to reproduce

Despite the problem is irregular and there is no 100% algorithm how to reproduce it, here are the steps allowing with more or less probability to get the problem:
1. Replace `navigation2/nav2_bringup/bringup/rviz/nav2_default_view.rviz` file with minimal repro-configuraiton attached to current issue: [nav2_default_view.rviz.zip](https://github.com/ros-visualization/rviz/files/6617714/nav2_default_view.rviz.zip)
2. Run Nav2 simulation as usual:
```
ros2 launch nav2_bringup tb3_simulation_launch.py
```
3. In appeared RViz screen set robot pose by "2D Pose Estimate"
4. enable-disable many times the "Controller" checkbox as shown in the picture below:
[image]
5. If problem won't appear within ~30-40 seconds, close RViz and repeat starting from step No.2.

### Environment
* Ubuntu 20.04 x64
* ROS2 Distro: Rolling built from sources (`ros2.repos` from Jun 1st).
* Nav2 main and Nav2 dependencies were [build from sources](https://navigation.ros.org/build_instructions/index.html#manually-build-main). Nav2 commit: https://github.com/ros-planning/navigation2/commit/b63b11f8a638faf389d6b99911d95899abe1580b
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
RViz commit: 94f6dd6958b2927a307893ff174f182dcc7afd0e
```
$ ros2 run rviz2 rviz2 -l
...
[INFO] [1623165832.507667729] [rviz2]: *-*-* OGRE Initialising
[INFO] [1623165832.507680536] [rviz2]: *-*-* Version 1.12.1 (Rhagorthua)
...
[INFO] [1623166434.886964038] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
```

Qt version is not printed by rviz:
```
$ qmake --version
Using Qt version 5.12.8 in /usr/lib/x86_64-linux-gnu
```
* If source build, which git commit?
The problem is long-standing detected since Crystal. Accused commit was not found.

SteveMacenski, 2021-06-08:
Can you boil down the tracebacks to areas of issue? I see some TF message filter areas mostly.

rhaschke, 2021-06-08:
Please file your issue over here: https://github.com/ros2/rviz. This repo is ROS1 only.

closed 2021-06-08 by rhaschke

AlexeyMerzlyakov, 2021-06-08:
Re-opened for RViz2 repo: https://github.com/ros2/rviz/issues/703