ros-visualization/rviz #1641: Rviz Freezes on "Save Config as"

lorepieri8, 2021-06-23:
Rviz Freezes when clicking Save Config as. In particular it becomes slightly grey and unresponsive. It  gets back to normal after about 1 minute, but without asking where to save the file.

Save Config is working fine.

### Your environment
* OS Version: e.g. Ubuntu 20.04
* ROS Distro: Noetic
*This is RViz version 1.14.8 (noetic).
Compiled against Qt version 5.12.8.
Compiled against OGRE version 1.9.0 (Ghadamon).

rhaschke, 2021-06-24:
Thanks for reporting this issue. I suspect #1639 for this, but I cannot reproduce the issue. Could you please provide more details regarding your specific display configuration, i.e. which plugins are active?

lorepieri8, 2021-06-24:
After a fresh start, I'm also not able to reproduce. The unresponsive period is very short (1 second max).
The plugins are HandEyeCalibration, MotionPlanning and Display (Trajectory, MotionPlanning, TF, Camera)

rhaschke, 2021-06-24:
Hm. If the issue isn't reproducible, I cannot debug it. Could you please try reproducing it with different setups?
What do you mean by _fresh start_?

rhaschke, 2021-06-24:
> The unresponsive period is very short (1 second max).

Maybe it's just taking time to initially load the shared libraries for the system dialogs?
Is there a difference after a fresh reboot and after the first successful execution of the dialog?

lorepieri8, 2021-06-24:
I was able to reproduce (sigh). By fresh start I mean putting the computer to sleep and restarting it. No, it's not a startup problem, quite the opposite. After a while the memory usage goes up and rviz becomes increasingly sluggish when saving. Just killing the processes and restarting doesn't seem to improve the situation. 
Rviz is not the only heavy process, but it's the heaviest one according to htop:

        ```
  PRI  NI  VIRT   RES   SHR S CPU% MEM%   TIME+  Command
  20   0 3388M  484M  177M R 210.  3.1  8:32.34 /opt/ros/noetic/lib/rviz/rviz -d ...
```
CPU usage is 210% for rviz (I'm running on a 8 cores machine).

rhaschke, 2021-06-24:
You state that memory usage increases over time. But, the numbers shown are fine for rviz. How fast does the memory grow?

- First, could you please provide your .rviz config?
- Can you narrow down the increasing memory usage to a specific plugin?
- Is your memory usage running out-of-bounds, such that the PC starts to swap?

lorepieri8, 2021-06-24:
It may not be the memory, unfortunately it's hard for me to reproduce this too.  I will try to share more info if I get this issue again.

This is the config:
[https://gist.github.com/lorepieri8/7ace191de7705a6f672b1a17945a2234](url)

rhaschke, 2021-06-25:
Did you observe this issue the first time now (after the 1.14.8 release) or did you see it before as well?

lorepieri8, 2021-06-25:
After the 1.14.8 update.

rhaschke, 2021-08-22:
Closing due to inactivity.

closed 2021-08-22 by rhaschke

rhaschke, 2021-08-22:
Just a few hours later, I ran into this issue as well: It seems to be related to a strong load of the main thread.
When reverting #1639, the issue is gone.