ros-visualization/rviz #1642: Wireframe meshes

daniel86, 2021-06-25:
Hello,
I think it would be valuable if one could toggle on/off wireframe rendering for markers displayed in RViz.
It can be done easily through the OGRE material.
It is useful for debugging geometry but also for creating schematic images.
Best,
Daniel

rhaschke, 2021-06-25:
Please note that rviz for ROS1 isn't actively developed anymore. New features are accepted only if provided by the community.
In this vein, I am looking forward to your PR :wink:
How would you like to toggle wireframe rendering? Globally, for individual marker displays, or for individual markers?
What should be the GUI interface for this?

daniel86, 2021-06-30:
Thanks for the response!
I was hoping to find it in some display panel as a global control. But it is also nice with meshes :)
I won't promise to, but maybe I'll have a look if it can be added trivially some day. Or maybe my group will finally move to ROS2 before.