ros-visualization/rviz #1643: qt.qpa.xcb: could not connect to display

Feynman132, 2021-06-28:
Hi

I am using trying to use ROS via WSL2 and Ubuntu 20.04. I have installed and launched VcXsrv. However, I failed to launch rviz and failed to run/launch turtlesim as well. I received the following returns:

"qt.qpa.xcb: could not connect to display
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, xcb.

Aborted"

Could you advise? Thank you

rhaschke, 2021-06-28:
I suggest following [these instructions](https://github.com/rhaschke/lecture/wiki/WSL-install). 
Also, googling for the particular error message (which Qt-related, not rviz-related), will point you to possible solutions.

closed 2021-06-28 by rhaschke

Feynman132, 2021-06-29:
Thanks Robert. The problem is solved.