ros-visualization/rviz #1646: Cannot mix incompatible Qt library (5.12.8) with this library (5.15.2)

mostafanorouzi, 2021-07-13:
Hello
When I use the package in pyside2 UI I encounter the following error. While when I use PyQt5 I do not encounter this error. 

[ INFO] [1626154697.760858437]: Stereo is NOT SUPPORTED
[ INFO] [1626154697.760896634]: OpenGL device: GeForce 940MX/PCIe/SSE2
[ INFO] [1626154697.760906774]: OpenGl version: 4.6 (GLSL 4.6).
Cannot mix incompatible Qt library (5.12.8) with this library (5.15.2)
Aborted (core dumped)

* OS Version: ubuntu 20.04
* ROS Distro:  Noetic

rhaschke, 2021-07-13:
You cannot mix different versions of the same library! The default system version in 20.04 is 5.12.8, which is also used by rviz.
Obviously, you installed another one (5.15) with your pyside installation. PyQt5 probably uses the system version and thus doesn't conflict.

closed 2021-07-13 by rhaschke

mostafanorouzi, 2021-07-13:
how can i solve this problem ?  i want to use pyside2 ...

rhaschke, 2021-07-13:
Install pyside2 from the Ubuntu system packages: `sudo apt install python3-pyside2.qtgui`

mostafanorouzi, 2021-07-13:
solved that problem but i get new error 

[ INFO] [1626158805.132215735]: Stereo is NOT SUPPORTED
[ INFO] [1626158805.132255610]: OpenGL device: GeForce 940MX/PCIe/SSE2
[ INFO] [1626158805.132267106]: OpenGl version: 4.6 (GLSL 4.6).
Traceback (most recent call last):
  File "myviz.py", line 161, in <module>
    myviz = MyViz()
  File "myviz.py", line 95, in __init__
    layout.addWidget( self.frame )
TypeError: 'PySide2.QtWidgets.QBoxLayout.addWidget' called with wrong argument types:
  PySide2.QtWidgets.QBoxLayout.addWidget(VisualizationFrame)
Supported signatures:
  PySide2.QtWidgets.QBoxLayout.addWidget(PySide2.QtWidgets.QWidget, int = 0, PySide2.QtCore.Qt.Alignment = Default(Qt.Alignment))
  PySide2.QtWidgets.QBoxLayout.addWidget(PySide2.QtWidgets.QWidget)
Exception ignored in: <built-in method _destroy of weakref object at 0x7f7c042be360>
SystemError: <built-in method _destroy of weakref object at 0x7f7c042be360> returned a result with an error set

rhaschke, 2021-07-13:
Of course, you need to install _all_ required pyside2 packages. Here, obviously, you are missing the `QtWidgets` wrapper.

mostafanorouzi, 2021-07-13:
thank you for your answers
I installed python3-pyside2.qtgui, python3-pyside2.qtcore , python3-pyside2.qtwidgets but problem not solved.
can you test pyside2 for myrviz.py to launch it ?

rhaschke, 2021-07-13:
I won't have time to look into that issue anytime soon. But, yes, if you leave your python script here, someone at some point can have a look. Better than this, is to provide the error message you get.
I just remember that rviz currently doesn't support `PySide`. There is an open PR https://github.com/ros-visualization/rviz/pull/1526 to fix this, but there is a fundamental problem with the wrapper-code generated by `shiboken`. Feel free to continue the PR.