ros-visualization/rviz #1653: librviz : Creating a MotionPlanning display in Noetic without crashing

julien-audet, 2021-08-05:
Hello,

I've been using librviz successfully for the past two years in my own Qt application for the visualization of a robot doing finishing tasks in ROS Kinetic with Ubuntu 16.04. Here's a code snippet of how I initialized librviz classes and added rviz::Display objects : 

```
    // Construct and lay out render panel.
    rviz_render_panel_ = new rviz::RenderPanel();

    // Next we initialize the main RViz classes.
    rviz_manager_ = new rviz::VisualizationManager(rviz_render_panel_);
    rviz_render_panel_->initialize(rviz_manager_->getSceneManager(), rviz_manager_);
    rviz_manager_->initialize();
    rviz_manager_->startUpdate();
    
    // Create a display for MotionPlanning
    moveit_display_ = rviz_manager_->createDisplay("moveit_rviz_plugin/MotionPlanning", "MotionPlanning", true);
```
This used to work very well, but I've recently migrated from Kinetic to Noetic, and now, the last line (createDisplay) crashes without any warning or error for "moveit_rviz_plugin/MotionPlanning". It still works with any other rviz::Display objects, but crashes everytime when using "MotionPlanning". If I use rViz directly, I can easily create a MotionPlanning display and do A to B trajectories. Interestingly enough, it doesn't crash if I disable the display (isEnabled -> false) :
```
    moveit_display_ = rviz_manager_->createDisplay("moveit_rviz_plugin/MotionPlanning", "MotionPlanning", false);
```
But it crashes as soon as I enable it : moveit_display__->setEnabled(true)

**So, my question is :** 

Is the MotionPlanning display object deprecated in the Noetic version of librviz (I wouldn't think so since it works in rViz)? 

### Your environment
* OS Version: Ubuntu 20.04
* ROS Distro: Noetic
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
[ INFO] [1628189178.265110573]: Stereo is NOT SUPPORTED
[ INFO] [1628189178.265156109]: OpenGL device: Mesa Intel(R) UHD Graphics (CML GT2)
[ INFO] [1628189178.265191473]: OpenGl version: 4.6 (GLSL 4.6) limited to GLSL 1.4 on Mesa system.

rhaschke, 2021-08-22:
Sorry for the late reply. I think the core problem is that you (and the librviz tutorial) don't use a window manager at all: 
The [`VisualizationManager` is initialized w/o a second argument](https://github.com/ros-visualization/visualization_tutorials/blob/141ebe6d3cd7213af6be26412e8d3818665c324c/librviz_tutorial/src/myviz.cpp). Thus, getWindowManager returns `NULL`. In order to display the associated widget panel you need to specify a window manager.
Have you ever seen the panel in Kinetic? I'm not yet sure how/why it worked in Kinetic...
I will file a fixup in MoveIt to avoid the segfault, though.

julien-audet, 2021-09-07:
Sorry for not getting back to you sooner, I was away for two weeks. And you're right, I don't use a window manager at all. Still confused with why it worked in Kinetic. Thank you for the fixup! I'll try it tonight and then close the issue.

closed 2021-09-08 by julien-audet