ros-visualization/rviz #1657: Crash when quickly toggling robot_description_property in RobotStateDisplay

simonschmeisser, 2021-08-24:
I have a `RobotStateDisplay` that I use as a preview for an URDF text editor. When clicking somewhat quickly on the reload button it crashes. The reload button basically changes the robot_description on the param server and then calls
```
    robot_description_property_->setValue(QString());
    robot_description_property_->setValue(DefaultParamName);
```
to trigger a reload. I saw crashes a while ago when a Qt Quick Controls bug would emit the signal of a button twice and thus called update very rapidly. But now a collegue showed me that simply semi-rapidly clicking also leads to a crash after a while.

The backtrace is unfortunately very non-descript:
```
1  tf2_ros::TransformListener::subscription_callback_impl(ros::MessageEvent<tf2_msgs::TFMessage_<std::allocator<void>> const> const&, bool)                              0x7fe0435e2b8b 
2  rviz::PropertyTreeWidget::propertyHiddenChanged                                                                                          property_tree_widget.cpp 126 0x7fe047c2d80d 
3  QObject::event(QEvent *)                                                                                                                                              0x7fe045be8fb2 
4  QWidget::event(QEvent *)                                                                                                                                              0x7fe0480ad273 
5  QFrame::event(QEvent *)                                                                                                                                               0x7fe04815012e 
6  QAbstractScrollArea::event(QEvent *)                                                                                                                                  0x7fe048152e63 
7  QAbstractItemView::event(QEvent *)                                                                                                                                    0x7fe0482c7d9b 
8  QApplicationPrivate::notify_helper(QObject *, QEvent *)                                                                                                               0x7fe04806ce0c 
9  QApplication::notify(QObject *, QEvent *)                                                                                                                             0x7fe048074360 
10 QCoreApplication::notifyInternal2(QObject *, QEvent *)                                                                                                                0x7fe045bb88d8 
11 QCoreApplicationPrivate::sendPostedEvents(QObject *, int, QThreadData *)                                                                                              0x7fe045bbb497 
12 ??                                                                                                                                                                    0x7fe045c16373 
13 g_main_context_dispatch                                                                                                                                               0x7fe033b76537 
14 ??                                                                                                                                                                    0x7fe033b76770 
15 g_main_context_iteration                                                                                                                                              0x7fe033b767fc 
16 QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>)                                                                                            0x7fe045c1598f 
17 QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>)                                                                                                               0x7fe045bb6b2a 
18 QCoreApplication::exec()                                                                                                                                              0x7fe045bbfd30 
19 main                                                                                                                                     main.cpp                 296 0x557cf43ce080 

```

basically it seems that hiding invalidates some parts of the property tree that are later accessed in the slot

### Your environment
* OS Version: Ubuntu 18.04
* ROS Distro:  Melodic
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
    ```
[ INFO] [1629822612.423814117]: rviz version 1.14.8
[ INFO] [1629822612.423847305]: compiled against Qt version 5.12.10
[ INFO] [1629822612.423854628]: compiled against OGRE version 1.9.0 (Ghadamon)

    ```
* If source build, which git commit? 679c2853

simonschmeisser, 2021-08-24:
I suspect that the `Property` gets deleted before the slot is called

simonschmeisser, 2021-08-25:
I'm still wondering if there is anything multi-threaded going on here? I can see that `PlanningSceneDisplay` uses `addBackgroundJob` for loading the robot model (and LinkModel) but for RobotStateDisplay it seems to be synchronous?

rhaschke, 2021-08-25:
I shortly looked into the code this morning and didn't see anything related to threading.
In any case, why don't you just trigger the slot `changedRobotDescription()`?

The backtrace doesn't make sense to me. `tf2_ros::TransformListener::subscription_callback_impl` isn't called from `propertyHiddenChanged()`:
https://github.com/ros-visualization/rviz/blob/2d4003dcd98f6e79eeed75963ae37fe149a96643/src/rviz/properties/property_tree_widget.cpp#L122-L128

Looks like you corrupted your stack beforehand. Did you run with `libasan` to check for any issue?

simonschmeisser, 2021-08-25:
`changedRobotDescription` is a private slot so that doesn't work but also calling
```
if (isEnabled())
  reset();
```
gives the same results. The stack trace did not contain the bogus tf2 this time, I agree that this is most like due to memory being corrupted already

I'll check out ASAN

simonschmeisser, 2022-12-22:
We changed some code path in our software apparently and can now quite easily reproduce this. The problem is that a queued connection is used here:
https://github.com/ros-visualization/rviz/blob/2d4003dcd98f6e79eeed75963ae37fe149a96643/src/rviz/properties/property_tree_widget.cpp#L113

and that the `Property*` becomes invalid in the mean time while waiting in the event loop to be executed.

I could not figure out why this is a delayed connection instead of a direct connection, it was changed in
https://github.com/ros-visualization/rviz/commit/3bbf90e1079f1976280a4f4720b2b9b174ad5583
with a commit message relating to `PointCloudDisplay`

It seems that the ui freezes while reloading the robot description if I make this a direct instead of a queued connection

simonschmeisser, 2022-12-22:
Ok, I was off with the previous it seems. Output of a succesful reload is
```
rviz::Property(0x5616e40308e0, name = "Links")
12
QModelIndex(3,0,0x5616e408f440,rviz::PropertyTreeModel(0x5616e407b4d0))
false
rviz::Property(0x5616e40308e0, name = "Links")
12
QModelIndex(3,0,0x5616e408f440,rviz::PropertyTreeModel(0x5616e407b4d0))
false
rviz::FloatProperty(0x5616e59b78e0, name = "Alpha")
0
QModelIndex(5,0,0x5616e5a5ccd0,rviz::PropertyTreeModel(0x5616e407b4d0))
true
rviz::FloatProperty(0x5616e59c90c0, name = "Alpha")
0
QModelIndex(8,0,0x5616e51667f0,rviz::PropertyTreeModel(0x5616e407b4d0))
true
rviz::FloatProperty(0x5616e50cde00, name = "Alpha")
0
QModelIndex(12,0,0x5616e6405830,rviz::PropertyTreeModel(0x5616e407b4d0))
true
rviz::FloatProperty(0x5616e5c04890, name = "Alpha")
0
QModelIndex(13,0,0x5616e5feb0c0,rviz::PropertyTreeModel(0x5616e407b4d0))
true
rviz::FloatProperty(0x5616e59ce600, name = "Alpha")
0
QModelIndex(15,0,0x5616e646d080,rviz::PropertyTreeModel(0x5616e407b4d0))
true
rviz::FloatProperty(0x5616e60f4170, name = "Alpha")
0
QModelIndex(28,0,0x5616e5bf58d0,rviz::PropertyTreeModel(0x5616e407b4d0))
true
rviz::FloatProperty(0x5616e64a3fd0, name = "Alpha")
0
QModelIndex(30,0,0x5616e6465670,rviz::PropertyTreeModel(0x5616e407b4d0))
true
```
when it crashes it goes only till after the second "Links". So I need to search for "Alpha" instead.
```
void PropertyTreeWidget::propertyHiddenChanged(const Property* property)
{
  if (model_)
  {
    qDebug() << property;
    qDebug() << property->rowNumberInParent();
    qDebug() << model_->parentIndex(property);
    qDebug() << property->getHidden();
    setRowHidden(property->rowNumberInParent(), model_->parentIndex(property), property->getHidden());
  }
}
```

PR #1795 (2023-05-18): call propertyHiddenChanged synchronously
Property might no longer exist when event is triggered

fixes #1657
closed 2023-05-18 by rhaschke