ros-visualization/rviz #1663: Marker basic colors from mesh resources (.dae) are not showing

SalahEddineGhamri, 2021-09-10:
While tying to use a custom marker in rviz by loading the mesh resources from a .dae file, the colors defined are not showing correctly in rviz environment. I used the frame.dae file provided in this repository follwing this link: 
https://github.com/ros-visualization/rviz/blob/22b81ecfea7ea7ed69e35d537abf6f50c8e5f1d7/src/test/meshes/frame.dae
Although ```marker.mesh_use_embedded_materials = true;```  is correctly set colors from the dae are showing white. I still can taint the mesh using ```marker.colors.r g b a``` but original material from dae are showing as white.

the results are showing in this image:
[image]

### Your environment
* OS Version: e.g. Ubuntu 16.04
* ROS Distro: [Kinetic ]
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
    ```
    his is RViz version 1.12.17 (kinetic).
    Compiled against Qt version 5.5.1.
    Compiled against OGRE version 1.9.0 (Ghadamon).
    ```

SalahEddineGhamri, 2021-09-10:
a workaround for this problem is to use texture files for defining the dae materials. For some reason by doing so colors are showing correctly. A good example to follow is the PR2 description dae file in this repository.

rhaschke, 2021-09-15:
I cannot confirm this issue on Melodic or Noetic. The [mesh_marker_test](https://github.com/ros-visualization/rviz/blob/noetic-devel/src/test/mesh_marker_test.cpp) produces the expected behaviour:
[image]

If that issue is only present in Kinetic, we won't fix that anymore. Kinetic ran [EOL in April 2021](http://wiki.ros.org/Distributions#List_of_Distributions). I suggest switching to a more recent version of rviz. If your stuck to Kinetic, you can still build a more recent rviz from source.