ros-visualization/rviz #1672: rosrun rviz rviz -> libGL error: No matching fbConfigs or visuals found; failed to load driver: swrast

MrZMN, 2021-09-30:
### My environment
* OS Version: Ubuntu 20.04
* ROS Distro: Noetic

When I run **rosrun rviz rviz**, the followings are the results:

[ INFO] [1633008890.254080982]: rviz version 1.14.9
[ INFO] [1633008890.254141864]: compiled against Qt version 5.12.8
[ INFO] [1633008890.254154791]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1633008890.265745958]: Forcing OpenGl version 0.
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
Segmentation fault (core dumped)

I have tried solutions e.g., adding LIBGL_ALWAYS_INDIRECT= to bashrc (cf. https://github.com/ros-visualization/rviz/issues/1522), but it doesn't work

I also have installed mesa-utils and libgl1-mesa-glx libraries, but it didn't help either.

The Ubuntu OS runs as a virtual machine on Parallels, though I think the error has no relation with it.

Any ideas? Many thanks!

MrZMN, 2021-10-01:
This has been resolved. This error is due to problem of libOpenGL library version, and only happens on Parallel Desktop VMs. This is solved by running:

**sudo mv /usr/lib/x86_64-linux-gnu/libOpenGL.so.0.0.0 /usr/lib/x86_64-linux-gnu/libOpenGL.so.0.0.0.orig**
and
**sudo ln -s /usr/lib/x86_64-linux-gnu/libGL.so.1 /usr/lib/x86_64-linux-gnu/libOpenGL.so.0.0.0 # Replace with libGL.so.1**

And this solved my problem.

closed 2021-10-01 by MrZMN

tingtingElec, 2021-10-19:
> This has been resolved. This error is due to problem of libOpenGL library version, and only happens on Parallel Desktop VMs. This is solved by running:
>
> **sudo mv [...]

Thank you so much ! I have been looking for this solution for more than one hour.Finally your answer solved my problem

ajielin, 2021-11-27:
Thank you so much !!! I works!