ros-visualization/rviz #1676: Wrong visual about PointStamped radius

Julien-Livet, 2021-11-02:
I observed a vocabulary mistake about PointStamped display : the field "Radius" currently matches to the diameter (with TF display).
I think it is more simple to rename the field than correct the code (div by 2)...

[image]
### Your environment
* OS Version: e.g. Xubuntu 20.04
* ROS Distro: [Noetic]
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
```[ INFO] [1635843067.996186502]: rviz version 1.14.10
[ INFO] [1635843067.996297233]: compiled against Qt version 5.12.8
[ INFO] [1635843067.996307080]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1635843068.006034471]: Forcing OpenGl version 0.
[ INFO] [1635843068.251274587]: Stereo is NOT SUPPORTED
[ INFO] [1635843068.251392943]: OpenGL device: llvmpipe (LLVM 12.0.0, 128 bits)
[ INFO] [1635843068.251541995]: OpenGl version: 3,1 (GLSL 1,4).```

rhaschke, 2021-11-02:
Well spotted!

rhaschke, 2021-11-03:
Fixed via #1678.

PR #1678 (2021-11-03): Fix scaling of PointVisual's sphere
Fixes #1676
closed 2021-11-03 by rhaschke