ros-visualization/rviz #1679: Problem with rviz_cube.skeleton File

DasSchilli, 2021-11-04:
I have an rviz plugin which can create a cube or sphere by clicking with the mouse on a symbol. The mesh files of my cube or spheres are the default ogre_media/models files of rviz pkg. Since the last update I got an error when I add my rviz::tool. 

`[ERROR] [1636065651.467587506]: Error retrieving file [file:///opt/ros/noetic/share/rviz/ogre_media/models/rviz_cube.skeleton]: Couldn't open file /opt/ros/noetic/share/rviz/ogre_media/models/`

Where can I find these files ?

### Your environment
* OS Version: e.g. Ubuntu 20.04
* ROS Distro: [Noetic]
* RViz, Qt, OGRE, OpenGl version as printed by rviz:

```[ INFO] [1636065240.114715632]: rviz version 1.14.10
[ INFO] [1636065240.114759044]: compiled against Qt version 5.12.8
[ INFO] [1636065240.114767030]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1636065240.122419459]: Forcing OpenGl version 0.
[ INFO] [1636065240.700930695]: Stereo is NOT SUPPORTED
[ INFO] [1636065240.700972114]: OpenGL device: NVIDIA GeForce RTX 2060/PCIe/SSE2
[ INFO] [1636065240.700986853]: OpenGl version: 4.6 (GLSL 4.6).```

rhaschke, 2021-11-05:
Loading of skeleton files was introduced in #1654. But that should be optional.
Could you provide a pointer to your repo to try to reproduce that issue? I haven't seen that for markers, for example.

rhaschke, 2021-11-05:
You can safely ignore this error message. Probably, it should be changed to an `info` or even `debug` message here:
https://github.com/ros-visualization/rviz/blob/3924f0e2500aec6cd532d97484a88e055aa2cac9/src/rviz/mesh_loader.cpp#L757
@kar-re, any objections?

DasSchilli, 2021-11-05:
> Loading of skeleton files was introduced in #1654. But that should be optional. Could you provide a pointer to your repo to try to reproduce that issue? I haven't seen that for markers, for example.

Thanks for the quick reply. Currently I have no repo. for this project, I will take care of this. 

Thanks for the explanation, now it is clear. I load the markers as a mesh to be able to scale them up or down in the 3D scene in RVIZ by using the mouse. After placing the sphere or cube has a fixed size and is published as a marker.

This line produces the error.

`rviz::loadMeshFromResource( resources_[curr_resource_] )`

DasSchilli, 2021-11-05:
> You can safely ignore this error message. Probably, it should be changed to an `info` or even `debug` message here:
> 
> https://github.com/ros-visualization/rviz/blob/3924f0e2500aec6cd532d97484a88e055aa2cac9/src/rviz/mesh_loader.cpp#L757
> 
> 
> @kar-re, any objections?

Ok thx for info.
If this were possible, it would be very helpful.

commit 45e5a9b1 (2021-11-05): Silent warning about .skeleton files not found

Fixes #1679
closed 2021-11-05 by rhaschke