ros-visualization/rviz #1680: Rviz segmentation fault

khaledelmadawi, 2021-11-14:
rosrun rviz rviz 
[ INFO] [1636925888.856839755]: rviz version 1.14.10
[ INFO] [1636925888.856881294]: compiled against Qt version 5.12.8
[ INFO] [1636925888.856891190]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1636925888.864509882]: Forcing OpenGl version 0.
Segmentation fault (core dumped)

glxinfo | grep "OpenGL"
OpenGL vendor string: Mesa/X.org
OpenGL renderer string: llvmpipe (LLVM 12.0.1, 256 bits)
OpenGL core profile version string: 4.5 (Core Profile) Mesa 21.2.5 - kisak-mesa PPA
OpenGL core profile shading language version string: 4.50
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL core profile extensions:
OpenGL version string: 3.1 Mesa 21.2.5 - kisak-mesa PPA
OpenGL shading language version string: 1.40
OpenGL context flags: (none)
OpenGL extensions:
OpenGL ES profile version string: OpenGL ES 3.2 Mesa 21.2.5 - kisak-mesa PPA
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
OpenGL ES profile extensions:

lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.3 LTS
Release:	20.04
Codename:	focal

can you please give a hand?

rhaschke, 2021-11-15:
I have seen multiple issues reported in the past involving the `llvmpipe` software renderer. I think you should try to fix your system to use your actual GPU hardware. Search Google for "llvmpipe used instead of ..."

rhaschke, 2021-11-15:
I'm afraid I cannot help remotely with this hardware issue. You should try to compile with debug symbols and debug yourself.
My suspicion is that the `llvmpipe` software renderer is not mature enough yet.

lprobsth, 2022-10-11:
I also have this error with melodic on arch linux. Noetic does not have this problem on the same machine.

I have examined the compile process very closely and there seemed to be no difference that should result in this error. Someone also recommended to use the noetic branch and compile it under melodic - but that did not help either.

The arch linux system:
```
# glxinfo | grep "OpenGL"
OpenGL vendor string: Intel
OpenGL renderer string: Mesa Intel(R) UHD Graphics 620 (KBL GT2)
OpenGL core profile version string: 4.6 (Core Profile) Mesa 22.1.7
OpenGL core profile shading language version string: 4.60
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL core profile extensions:
OpenGL version string: 4.6 (Compatibility Profile) Mesa 22.1.7
OpenGL shading language version string: 4.60
OpenGL context flags: (none)
OpenGL profile mask: compatibility profile
OpenGL extensions:
OpenGL ES profile version string: OpenGL ES 3.2 Mesa 22.1.7
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
OpenGL ES profile extensions:
```

Another Ubuntu 20 machine where rviz is not working:
```
OpenGL vendor string: NVIDIA Corporation
OpenGL renderer string: NVIDIA GeForce GTX 1660 Ti with Max-Q Design/PCIe/SSE2
OpenGL core profile version string: 4.6.0 NVIDIA 510.85.02
OpenGL core profile shading language version string: 4.60 NVIDIA
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL core profile extensions:
OpenGL version string: 4.6.0 NVIDIA 510.85.02
OpenGL shading language version string: 4.60 NVIDIA
OpenGL context flags: (none)
OpenGL profile mask: (none)
OpenGL extensions:
OpenGL ES profile version string: OpenGL ES 3.2 NVIDIA 510.85.02
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
OpenGL ES profile extensions:
```

An ubuntu 18 machine:
```
OpenGL vendor string: NVIDIA Corporation
OpenGL renderer string: NVIDIA GeForce GTX 1660 Ti with Max-Q Design/PCIe/SSE2
OpenGL core profile version string: 4.6.0 NVIDIA 510.47.03
OpenGL core profile shading language version string: 4.60 NVIDIA
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL core profile extensions:
OpenGL version string: 4.6.0 NVIDIA 510.47.03
OpenGL shading language version string: 4.60 NVIDIA
OpenGL context flags: (none)
OpenGL profile mask: (none)
OpenGL extensions:
OpenGL ES profile version string: OpenGL ES 3.2 NVIDIA 510.47.03
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
OpenGL ES profile extensions:
```

I did not compile rviz for melodic on the ubuntu 18 machine yet - but I suspect that this has nothing to do with the OpenGL version itself or the used GPU hardware.

rhaschke, 2022-10-11:
@lprobsth, you report different Linux distros as well as different GPU / Mesa uses. However, it's not clear at all, which of these configs work and which don't. To yield meaningful results you should only change a single variable always, i.e. GPU vs. Mesa or the linux distro, or the rviz version.
If the noetic version of rviz is running on your target system, I suggest compiling the noetic-devel branch of rviz under melodic (if you are stuck to melodic). This should yield the same result then. If not (as you report), you probably didn't correctly source your custom workspace.

antbono, 2023-09-07:
Hello everyone! 
When trying to [install ros2 rolling via source on ubuntu 22.04 ](https://docs.ros.org/en/rolling/Installation/Alternatives/Ubuntu-Development-Setup.html) I get a segmentation fault for the package `rviz_default_plugins`. 
The output of the compiling command is reported in this file
 
[rviz_default_plugins.txt](https://github.com/ros-visualization/rviz/files/12550619/rviz_default_plugins.txt)

Can anyone give me a hint?
Thanks!

antbono, 2023-09-07:
I solved using a single thread with the commmand

`MAKEFLAGS=-j1 colcon build --executor sequential --packages-select rviz_default_plugins`