ros-visualization/rviz #1683: The problem of deconstruction order for Class VisualizationFrame. Wild pointers cause the program to crash

Luwenyong, 2021-11-24:
I developed a panel plug-in using rviz. The plug-in uses the interface during initialization and creates the corresponding ogre:: scenenode.
When I deconstruct my plug-in, I need to destroy the corresponding model node.  use sceneManager in vis_manager_ interface destroySceneNode.

But when I close rviz, the core dump will be generated in rviz. 
    Segmentation fault (core dumped)

I checked the source code of rviz. If modify code here, it could be solve.
    rviz\src\rviz\src\rviz\visualization_frame.cpp
(https://github.com/ros-visualization/rviz/blob/melodic-devel/src/rviz/visualization_frame.cpp)
      VisualizationFrame::~VisualizationFrame()
      {
        for (int i = 0; i < custom_panels_.size(); i++)
        {
          delete custom_panels_[i].dock;
        }
      
        delete render_panel_;
        render_panel_ = nullptr;
     
        delete manager_;
        manager_ = nullptr;
     
        //for (int i = 0; i < custom_panels_.size(); i++)
        //{
        //  delete custom_panels_[i].dock;
        //}
      
        delete panel_factory_;
        panel_factory_ = nullptr;
      }

If there is any better way to fix it, maybe we could have a discuss here.

Environment
OS Version: Ubuntu 18.04
ROS Distro: Melodic
RViz, Qt, OGRE, OpenGl version as printed by rviz:

    [ INFO] [1637744810.930629000]: rviz version 1.13.18
    [ INFO] [1637744810.930694100]: compiled against Qt version 5.9.5
    [ INFO] [1637744810.930725100]: compiled against OGRE version 1.9.0 (Ghadamon)
    [ INFO] [1637744810.933829600]: Forcing OpenGl version 0.
    [ INFO] [1637744811.026552200]: Stereo is NOT SUPPORTED
    [ INFO] [1637744811.026656000]: OpenGL device: llvmpipe (LLVM 10.0.0, 256 bits)
    [ INFO] [1637744811.026704200]: OpenGl version: 3.1 (GLSL 1.4).

rhaschke, 2021-11-26:
Did you analyze the backtrace of your segfault? Is it accessing the already deleted scene graph? Could you provide a link to your source or - even better - a minimal toy example demonstrating the segfault?
Deleting the panels before the render panel seems to be reasonable. I'm not yet sure though, why the current implementation doesn't pose issues in general.

Luwenyong, 2021-11-29:
My plug-in is written like this：

class NodeManager : rviz::Panel
{
public:
    ...
    NodeManager();
    ~NodeManager();
    void onInitialize() override;
    ...
private:
    ...
    Ogre::SceneManager *sceneManager { nullptr };
    Ogre::SceneNode *node { nullptr };
}

void NodeManager::onInitialize()
{
    ...
    const string resource("./model.dae");
    rviz::loadMeshFromResource(resource);
    sceneManager = vis_manager_->getSceneManager();
    Ogre::Entity *entity = sceneManager->createEntity(resource);
    Ogre::SceneNode *root = sceneManager->getRootSceneNode();
    node = root->createChildSceneNode();
    node->attachObject(entity);
    node->setVisible(true);
    ...
}

NodeManager::~NodeManager()
{
    ...
    if (sceneManager != nullptr) {
        sceneManager->destroySceneNode(node);  // **Segmentation fault (core dumped)**
    }
    ...
}

Luwenyong, 2021-12-10:
Dear, rhaschke
     If you have time, please pay more attention to this question. Thank you very much.

rhaschke, 2021-12-10:
I'm waiting for _you_ to provide a _compiling_ minimal code example. Also, I asked for a backtrace. Thanks.

rhaschke, 2021-12-21:
Hopefully fixed via b0b5c34f3277d10fc368e5a1f0c91f7ee4f09377.

closed 2021-12-21 by rhaschke