ros-visualization/rviz #1688: Meshes rendering in the wrong orientation

civerachb-cpr, 2021-12-15:
It appears that there's a bug in Rviz 1.13.22 where links with STL or DAE visuals are being rendered incorrectly  The underlying links are correctly located, but the visual and collision geometry is a mess.

For example, here is the [Clearpath Warthog](https://github.com/warthog-cpr/warthog) model viewed using the command `roslaunch warthog_viz view_robot.launch` in Rviz 1.13.18:

[image]

The same, but with the Rviz version information
[image]

Here's the exact same robot, rendered in Rviz 1.13.22:
[image]

The robot renders correctly inside Gazebo 9.19.0:
[image]

but adding the RobotModel vizualization to Rviz results in the same mal-formed visual as above; the meshes are all oriented incorrectly in Rviz, even though Gazebo is publishing the TF and joint states correctly.

This also affects simpler models, like the [VLP-16 from velodyne_description](http://wiki.ros.org/velodyne_description)
[image]

You can see the Velodyne's base link and sensor origin frames are oriented vertically relative to each other, but the mesh itself is rotated off to the side.  The above was created with this included in the URDF for the robot:

```
  <xacro:include filename="$(find velodyne_description)/urdf/VLP-16.urdf.xacro" />
  <xacro:VLP-16 parent="front_mount" topic="velodyne_points">
    <origin xyz="0 0 0" rpy="0 0 0" />
  </xacro:VLP-16>
```

### environment
* OS Version: Ubuntu 18.04
* ROS Distro: Melodic
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
    ```
    [ INFO] [1639587205.192206801]: rviz version 1.13.22
    [ INFO] [1639587205.192246967]: compiled against Qt version 5.9.5
    [ INFO] [1639587205.192257016]: compiled against OGRE version 1.9.0 (Ghadamon)
    ```
* If source build, which git commit? n/a -- installed via .deb from official ROS apt sources
* System locale, i.e. the output of `echo "$LANG $LC_NUMERIC"`: `en_US.UTF8`

_Before reporting a rendering issue, try running RViz with `LANG=C rviz`!_
I tried this, the result was the same; the meshes were still rotated incorrectly when adding the RobotModel display type.

civerachb-cpr, 2021-12-15:
(I just received a report from a co-worker that Rviz 1.19.21 appears to render properly; the issue seems to be specifically with the 1.19.22 release. Hopefully that helps narrow it down.)

rhaschke, 2021-12-15:
Thanks for reporting! I will have a look asap.

rhaschke, 2021-12-15:
@clalancette + @sloretz: Please hold back any planned release sync (if any) due to this regression.

rhaschke, 2021-12-15:
The culprit is probably https://github.com/ros-visualization/rviz/pull/1685. But I need to understand why it happens for your meshes and not for the ones I tested with...

civerachb-cpr, 2021-12-15:
I'm confused too; in the example of the Warthog, the wheels appear to be oriented correctly, but the large yellow fender assemblies and the central chassis are not.

rhaschke, 2021-12-16:
@civerachb-cpr: I filed a fixup. Could you please verify that this fixes the issue for you?

knorth55, 2021-12-16:
I have the same issue.
[image]

rhaschke, 2021-12-16:
Fixed via #1689.

closed 2021-12-16 by rhaschke

knorth55, 2021-12-17:
now `rviz` `1.13.23` is released in `ros-testing`, and I update `rviz` with `apt`.
It works well, thank you for the quick fix.

[image]