ros-visualization/rviz #1690: Rviz  can not open

Hardliu, 2021-12-24:
ubuntu18.04 melodic
when I try to open rviz, it's will display errors.

[ INFO] [1640269957.506665159]: rviz version 1.13.23
[ INFO] [1640269957.506734274]: compiled against Qt version 5.9.5
[ INFO] [1640269957.506749333]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1640269957.514238673]: Forcing OpenGl version 0.

rviz::RenderSystem: error creating render window: boost::thread_resource_error: Resource temporarily unavailable
[ WARN] [1640269959.982329944]: OGRE EXCEPTION(2:InvalidParametersException): Window with name 'OgreWindow(0)' already exists in GLRenderSystem::_createRenderWindow at /build/ogre-1.9-B6QkmW/ogre-1.9-1.9.0+dfsg1/RenderSystems/GL/src/OgreGLRenderSystem.cpp (line 1057)

rviz::RenderSystem: error creating render window: OGRE EXCEPTION(3:RenderingAPIException): Unable to create a suitable GLXContext in GLXContext::GLXContext at /build/ogre-1.9-B6QkmW/ogre-1.9-1.9.0+dfsg1/RenderSystems/GL/src/GLX/OgreGLXContext.cpp (line 61)
[ERROR] [1640269960.307447341]: Unable to create the rendering window after 100 tries.
Segmentation fault

Hardliu, 2021-12-25:
> ubuntu18.04 melodic when I try to open rviz, it's will display errors.
> 
> [ INFO] [1640269957.506665159]: rviz version 1.13.23 [ INFO] [1640269957.506734274]: compiled against Qt version 5.9.5 [ INFO] [1640269957.506749333]: compiled against OGRE version 1.9.0 (Ghadamon) [ INFO] [1640269957.514238673]: Forcing OpenGl version 0.
> 
> rviz::RenderSystem: error creating render window: boost::thread_resource_error: Resource temporarily unavailable [ WARN] [1640269959.982329944]: OGRE EXCEPTION(2:InvalidParametersException): Window with name 'OgreWindow(0)' already exists in GLRenderSystem::_createRenderWindow at /build/ogre-1.9-B6QkmW/ogre-1.9-1.9.0+dfsg1/RenderSystems/GL/src/OgreGLRenderSystem.cpp (line 1057)
> 
> rviz::RenderSystem: error creating render window: OGRE EXCEPTION(3:RenderingAPIException): Unable to create a suitable GLXContext in GLXContext::GLXContext at /build/ogre-1.9-B6QkmW/ogre-1.9-1.9.0+dfsg1/RenderSystems/GL/src/GLX/OgreGLXContext.cpp (line 61) [ERROR] [1640269960.307447341]: Unable to create the rendering window after 100 tries. ++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++ glxinfo | grep "OpenGL"
> 
> OpenGL vendor string: NVIDIA Corporation OpenGL renderer string: NVIDIA GeForce RTX 3080 Ti/PCIe/SSE2 OpenGL core profile version string: 4.6.0 NVIDIA 495.29.05 OpenGL core profile shading language version string: 4.60 NVIDIA OpenGL core profile context flags: (none) OpenGL core profile profile mask: core profile OpenGL core profile extensions: OpenGL version string: 4.6.0 NVIDIA 495.29.05 OpenGL shading language version string: 4.60 NVIDIA OpenGL context flags: (none) OpenGL profile mask: (none) OpenGL extensions: OpenGL ES profile version string: OpenGL ES 3.2 NVIDIA 495.29.05 OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20 OpenGL ES profile extensions:

rhaschke, 2021-12-27:
I cannot reproduce this issue. Could you try different versions of rviz? Do standard GL apps, e.g. `glxgears`, work without issues?

Hardliu, 2021-12-27:
> I cannot reproduce this issue. Could you try different versions of rviz? Do standard GL apps, e.g. `glxgears`, work without issues?

Thank you for your reply，glxgears works well on my computer。
Actually, I dont konw the different versions of rviz, I install it by ROS，do you mean to install another version of rviz and compiles it?

Hardliu, 2021-12-27:
> > I cannot reproduce this issue. Could you try different versions of rviz? Do standard GL apps, e.g. `glxgears`, work without issues?
> 
> Thank you for your reply，glxgears works well on my computer。 Actually, I dont konw the different versions of rviz, I install it by ROS，do you mean to install another version of rviz and compiles it? By the way, gazeo can not open either.

Hi, I try to the version of kinetic-devel , it's has same error

rhaschke, 2021-12-27:
As I don't have your hardware setup available for testing, I'm afraid you need to compile and debug rviz yourself to narrow down the issue. Please compile the latest noetic-devel version using debug symbols and try debugging `RenderSystem::tryMakeRenderWindow`:
https://github.com/ros-visualization/rviz/blob/3924f0e2500aec6cd532d97484a88e055aa2cac9/src/rviz/ogre_helpers/render_system.cpp#L497

Try commenting out `both` `XSetErrorHandler` calls and report which X11 error you receive:
https://github.com/ros-visualization/rviz/blob/3924f0e2500aec6cd532d97484a88e055aa2cac9/src/rviz/ogre_helpers/render_system.cpp#L490
https://github.com/ros-visualization/rviz/blob/3924f0e2500aec6cd532d97484a88e055aa2cac9/src/rviz/ogre_helpers/render_system.cpp#L516

You could also try a newer version of Ogre, e.g. 1.12, compiled from source.

Hardliu, 2021-12-28:
> Ogre
Thank you for you reply, my computer is ubuntu18.04, ROS melodic, so i can't use noetic-devel:
++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
src/rviz/src/rviz/default_plugin/interactive_marker_display.cpp:110:100: error: no matching function for call to ‘interactive_markers::InteractiveMarkerClient::InteractiveMarkerClient(std::__shared_ptr_access<tf2_ros::Buffer, (__gnu_cxx::_Lock_policy)2, false, false>::element_type&, std::__cxx11::string)’
+++++++++++++++++++++++++++++++++++++++++++++++++++++++
Is there any other way to solve the problem of different version incompatitiy? Is there any the patch on noetic version ?

Hardliu, 2021-12-28:
> 

I try to build ogre1.12.3, but xviz build use orgre 1.9, how to replace it？

rhaschke, 2021-12-28:
To resolve the Melodic incompatibility, just include the noetic-devel version of [interactive_markers](https://github.com/ros-visualization/interactive_markers) into your catkin workspace.
To build rviz against another Ogre version, pass `-DOGRE_DIR=<your ogre install path>` as a cmake argument to your catkin build. When using catkin tools, that's done via: `catkin config --cmake-args DOGRE_DIR=`

closed 2022-01-04 by Hardliu

rhaschke, 2022-01-04:
Could you resolve your issue? If so, please shortly describe how.

haofengsiji, 2025-03-18:
finally, I found the bug source, https://github.com/OGRECave/ogre/issues/1057, it is because bug in orge since v1.70.