ros-visualization/rviz #1694: Feature request: robot_description topic

Timple, 2022-01-11:
To ease the transition to ROS2 we would like to use an `robot_description` topic instead of a parameter. 

Would a PR get accepted which expands the model view with a topic option?

Implementation would be an enum box which chooses between Parameter (default) or Topic.

130s, 2023-05-23:
Is setting robot description ROS Parameter on ROS2 side not an option? Just wondering, but in one of a ros1-2 bridge projects I think I saw that (robot running on ros1, then there's a bridge to receive some in ros2, but in ros2 redundantly set robot description). It's not nice but worked. Just fyi.

Timple, 2023-07-27:
I've created a plugin which solves this issue: https://github.com/nobleo/rviz_robot_description_topic

closed 2023-07-27 by Timple