ros-visualization/rviz #1699: RViz may crash when TF's childFrameId is " "

qinshichao, 2022-01-27:
When the childframeid of TF is ' ' and subscribed, RViz will crash.
[image]

When the childframeid of TF is '\a'  or '\e' or other strings starting with' \ ',Abnormal display of RViz.
[image]

### Your environment
* OS Version: e.g. Ubuntu 18.04
* ROS Distro: [Melodic]
* RViz, Qt, OGRE, OpenGl version as printed by rviz:

[image]

rhaschke, 2022-01-27:
I cannot confirm these issues:
TF already rejects/ignores an empty child transform with the following error message on console:
`[ERROR] ros.rosconsole_bridge.TF_NO_CHILD_FRAME_ID: Ignoring transform from authority "unknown_publisher" because child_frame_id not set`

When sending a child frame with a non-printable control character in its name (e.g. `\a`), what exactly do you expect to happen?

qinshichao, 2022-01-27:
> I cannot confirm these issues: TF already rejects/ignores an empty child transform with the following error message on console: `[ERROR] ros.rosconsole_bridge.TF_NO_CHILD_FRAME_ID: Ignoring [...]

It's not empty, it's a space.
When sending a child frame with a non-printable control character in its name (e.g. \a), I expect it to remain '\ a' without escape.

qinshichao, 2022-01-27:
> To have a frame name like `\a` you obviously need to escape the backslash like so: `"\\a"
Can string escape be prevented in code?

rhaschke, 2022-01-27:
> To have a frame name like `\a` you obviously need to escape the backslash like so: `"\a"

> > Can string escape be prevented in code?

No. That's about interpretation of chars in your programming language:
-  `\a` is interpreted as a control character
- `\\a` is interpreted as backslash + letter a

rhaschke, 2022-01-27:
crash fixed via #1700.

PR #1700 (2022-01-27): MovableText: gracefully handle string of whitespaces
Fixes #1699
closed 2022-01-27 by rhaschke