ros-visualization/rviz #1702: Mesh with normal have no light in latest release 1.14.13 (regression since 1.14.11)

xbroquer, 2022-01-27:
Hello @rhaschke ,

Affect version : ROS Noetic , ubuntu 20.04
RViz 1.14.13 installed using ubuntu apt upgrade on Jan 27th 2022

In my ROS setup I display some meshes generated using blender. After rviz upgrade the RVIZ 3D scene  seems to have no light.

I cloned rviz in my catkin workspace and checked out tag 1.14.11: no issue

I compared the changes between 1.14.11 and HEAD of noetic-devel, I found some changes in mesh_loader.
After investigation I found that you removed the transformation of normals.

By re-applying the transformation on normal , the mesh well rendered.

My local revert:
[image]

with lastest release:

[image]

With the patch:

[image]

rhaschke, 2022-01-28:
Could you please provide an example mesh file exhibiting this behavior?

rhaschke, 2022-01-28:
I believe that the complicated transform is not needed, because it just rescales the normals, which is subsequently undone by normalizing the vectors. However, it looks like I removed the transform completely. I think #1703 should already solve the issue. Could you please verify? Please, provide an example mesh in any case. Thanks.

xbroquer, 2022-01-28:
Thanks for your quick reply, I am not allowed to attach the small car, but I attached a truck, I am testing #1703 and come back to you asap

xbroquer, 2022-01-28:
[image]

[image]

rhaschke, 2022-01-28:
I couldn't find the truck mesh file - there was no attachment. But from your pictures, I conclude that my simpler fix resolves the issue for you?

xbroquer, 2022-01-28:
no, in my previous comment , the first screenshot is with the "simple" fix. 
I will add the mesh in my https://github.com/xbroquer/rviz_collada_marker this afternoon along with launch + rviz conf file

xbroquer, 2022-01-28:
Can you please try: roslaunch rviz_collada_marker test_truck.launch from https://github.com/xbroquer/rviz_collada_marker

xbroquer, 2022-01-28:
Using #1703:
[image]
It's not cpompletely dark when moving the rviz camera:
[image]

Here using  https://github.com/xbroquer/rviz/commit/3ac6cf41eb3ccd2cd769e775557f44fe6e177196
[image]

rhaschke, 2022-01-28:
I fixed the issue in #1703. Thanks for providing the mesh file. This helped to debug this issue.

xbroquer, 2022-01-28:
Thanks! Do you know when I should expect this fix released and available on deb repo for Noetic ?

rhaschke, 2022-01-28:
I will prepare a release within next week latest.
However, both Melodic and Noetic just recently had a sync: https://discourse.ros.org/c/release/16. Thus, it will take a month or so until the next release will be available. Consider using the debian packages from the [ROS `testing` repo](http://wiki.ros.org/TestingRepository) meanwhile.

rhaschke, 2022-01-28:
Fixed via #1703

PR #1703 (2022-01-28): mesh loader: re-add transform of mesh normals
Hopefully fixes #1702.
closed 2022-01-28 by rhaschke