ros-visualization/rviz #1706: UR10e mesh files have black colour in rviz

AminSeffo, 2022-01-31:
### My environment
* OS Version: Ubuntu 20.04.3 LTS
* ROS Distro: [Noetic]
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
```
[ INFO] [1643653127.450372931]: rviz version 1.14.13
[ INFO] [1643653127.450441175]: compiled against Qt version 5.12.8
[ INFO] [1643653127.450463892]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1643653127.466547213]: Forcing OpenGl version 0.
[ INFO] [1643653128.141746527]: Stereo is NOT SUPPORTED
[ INFO] [1643653128.141801451]: OpenGL device: NVIDIA GeForce RTX 3090/PCIe/SSE2
[ INFO] [1643653128.141817482]: OpenGl version: 4,6 (GLSL 4,6).
[ INFO] [1643653137.045036710]: Loading robot model 'ur10e_robot'...
```
Hello everyone,

I have recently just updated my Ubuntu with  `sudo apt update && sudo apt upgrade` and then I got lighting problems and  my UR10e meshes are looking black in RVIZ as you can see in the screenshot below.

I didn't edit anything in robot description file URDF.

Any help is very appreciated.

[image]

rhaschke, 2022-01-31:
Duplicate of #1702, fixed in #1703. New release pending.

AminSeffo, 2022-01-31:
Hey @rhaschke 
thank you very mush for your response, but how can I re-add transform of mesh normals?
I installed **ros noetic desktop** and I have no clue how to edit ros files like that.

AminSeffo, 2022-01-31:
I cloned Rviz 1.14.11 in my catkin_ws like #1702 and it works again.
Thank you for your help.

rhaschke, 2022-02-12:
Fixed via #1703

closed 2022-02-12 by rhaschke