ros-visualization/rviz #1708: Dependency of yaml-cpp library

captain-yoshi, 2022-02-04:
This is not a big deal, but in my project I use the `SendFilePath.srv` message to reload RViz for viewing different robots trajectories.

I don't want my project to depend on RViz just for the msg service, because I need a specific version of `yaml-cpp` which is also a dependency of RViz. Conflicts occurs because RViz install automatically installs `libyaml-cpp` when installing from Ubuntu package system (the versions are different).

In the future I would also like to use `rviz_visual_tools` but don't need the [gui](https://github.com/PickNikRobotics/rviz_visual_tools/blob/master/CMakeLists.txt#L90-L93) part of the project that depends on RViz. Because of the way the package is created, am I stuck with the `rviz_visual_tools_gui` when depending on rviz_visual_tools ?

I think my only options are:
- Build RViz from source so that RViz and my project uses the same yaml-cpp library. Not really a viable option. 
- Incorporate the yaml-cpp library into my project (directly or via git-submodules) and change the namespace. As I target my project to build into melodic and noetic I think this is the better choice.
- Build `SendFilePath.srv` in my project. Still can't use rviz_visual_tools.

Is there other options ?

https://user-images.githubusercontent.com/32679594/152556688-f0fdb126-4435-41e8-a499-f4f16fe04ba6.mp4

rhaschke, 2022-02-04:
If you want to release your project into the ROS ecosystem you should indeed find a way to coexist with the system `yaml-cpp`.
However, runtime issues only arise when an executable links multiple _different_ versions of a library. But `librviz_visual_tools.so` doesn't link against `librviz` or `libyaml-cpp`:
`ldd /opt/ros/noetic/lib/librviz_visual_tools.so|grep yaml  -> no results`
Thus, the 3rd option is feasible as well (for now - until maintainers of `rviz_visual_tools` decide to include functionality from `librviz`).

closed 2022-02-04 by rhaschke

captain-yoshi, 2022-02-04:
Super !

I was using `libmoveit_visual_tools` and thought that it was `librviz_visual_tools`.

```bash
$ ldd devel/lib/libmoveit_visual_tools.so | grep rviz

librviz_visual_tools.so.3.9.1 => /opt/ros/noetic/lib/librviz_visual_tools.so.3.9.1 (0x00007fe093784000)
librviz_visual_tools_gui.so.3.9.1 => /opt/ros/noetic/lib/librviz_visual_tools_gui.so.3.9.1 (0x00007fe093752000)
librviz_visual_tools_remote_control.so.3.9.1 => /opt/ros/noetic/lib/librviz_visual_tools_remote_control.so.3.9.1 (0x00007fe092ad2000)
librviz_default_plugin.so => /opt/ros/noetic/lib/librviz_default_plugin.so (0x00007fe092520000)
librviz.so => /opt/ros/noetic/lib/librviz.so (0x00007fe091c0f000)

```

rhaschke, 2022-02-04:
In this case, you are stuck to option 2.