ros-visualization/rviz #1713: Cmake problem

AaronWaxson, 2022-02-11:
I have a rviz_plugin project, that's used catkin_make build it and can add it in rviz.  
But now I have a new need which put the rviz_plugin project in a cmake project. And I used cmake build all project successfully, but when I use the plugin in rviz, it give a bug back.
Bug info like that:  
`The class required for this panel, 'graph_rviz_plugin/Line graph', could not be loaded.
Error:
Could not find library corresponding to plugin graph_rviz_plugin/Line graph. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.`  
How can I resolve it? Please give some advices for me, thanks!

### Environment
* OS Version: Ubuntu 20.04
* ROS Distro: [Noetic]

rhaschke, 2022-02-11:
This question should be better asked at https://answers.ros.org. Please, post real issues and bug reports here only.
Did you check that rospack finds your plugin: https://answers.ros.org/question/298919/rospack-can-not-find-my-plugin/

closed 2022-02-11 by rhaschke

AaronWaxson, 2022-02-11:
Sorry and thank you.