ros-visualization/rviz #1718: PointCloud errors and crash with Ogre 1.12.13

simonschmeisser, 2022-02-21:
I'm running 92d83c3c2ccb87c3bc576ff26e1c9b1eff4f6a71 (with the mesa hack disabled locally) and when adding a `PlanningSceneDisplay` I get lot's of the following ROS_ERRORs when publishing a PointCloud to be displayed as Octomap:
```
[ERROR] [1645434400.287826510]: No techniques available for material [PointCloudMaterial1636Sphere]
[ERROR] [1645434400.287877485]: No techniques available for material [PointCloudMaterial1636Box]
[ERROR] [1645434400.288116566]: No techniques available for material [PointCloudMaterial1637Sphere]
[ERROR] [1645434400.288142249]: No techniques available for material [PointCloudMaterial1637Box]
[ERROR] [1645434400.288405009]: No techniques available for material [PointCloudMaterial1638Sphere]
[ERROR] [1645434400.288440468]: No techniques available for material [PointCloudMaterial1638Box]
[ERROR] [1645434400.288644570]: No techniques available for material [PointCloudMaterial1639Sphere]
[ERROR] [1645434400.288665005]: No techniques available for material [PointCloudMaterial1639Box]
[ERROR] [1645434400.288852429]: No techniques available for material [PointCloudMaterial1640Sphere]
[ERROR] [1645434400.288873083]: No techniques available for material [PointCloudMaterial1640Box]
[ERROR] [1645434400.289058814]: No techniques available for material [PointCloudMaterial1641Sphere]
[ERROR] [1645434400.289077285]: No techniques available for material [PointCloudMaterial1641Box]
[ERROR] [1645434400.289266442]: No techniques available for material [PointCloudMaterial1642Sphere]
[ERROR] [1645434400.289287367]: No techniques available for material [PointCloudMaterial1642Box]
[ERROR] [1645434400.289468341]: No techniques available for material [PointCloudMaterial1643Sphere]
[ERROR] [1645434400.289489797]: No techniques available for material [PointCloudMaterial1643Box]
[ERROR] [1645434400.289666212]: No techniques available for material [PointCloudMaterial1644Sphere]
[ERROR] [1645434400.289684352]: No techniques available for material [PointCloudMaterial1644Box]
[ERROR] [1645434400.289858414]: No techniques available for material [PointCloudMaterial1645Sphere]
[ERROR] [1645434400.289877376]: No techniques available for material [PointCloudMaterial1645Box]
[ERROR] [1645434400.290049935]: No techniques available for material [PointCloudMaterial1646Sphere]
[ERROR] [1645434400.290068386]: No techniques available for material [PointCloudMaterial1646Box]
[ERROR] [1645434400.290243419]: No techniques available for material [PointCloudMaterial1647Sphere]
[ERROR] [1645434400.290262521]: No techniques available for material [PointCloudMaterial1647Box]
[ERROR] [1645434400.290432035]: No techniques available for material [PointCloudMaterial1648Sphere]
```

The octomap is not visible.

When scrolling in the visualization I get a crash with the following backtrace: https://gist.github.com/simonschmeisser/86f8cdf6a482d4f6e5e9a1e1793ec141

[image]

### Your environment
* OS Version: e.g. Ubuntu 20.04
* ROS Distro:  Noetic
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
    ```
    [ INFO] [1645435522.659729408]: rviz version 1.14.14
[ INFO] [1645435522.659793677]: compiled against Qt version 5.12.8
[ INFO] [1645435522.659811711]: compiled against OGRE version 1.12.13 (Rhagorthua)
[ INFO] [1645435522.676000365]: Forcing OpenGl version 0.
[ INFO] [1645435522.993456542]: Stereo is NOT SUPPORTED
[ INFO] [1645435522.993594178]: OpenGL device: AMD RENOIR (DRM 3.41.0, 5.13.0-30-generic, LLVM 12.0.0)
[ INFO] [1645435522.993647246]: OpenGl version: 4,6 (GLSL 4,6).
[ERROR] [1645435523.005945791]: Error: ScriptCompiler - unexpected token in ogre1.9.fontdef(1): 'Liberation'. If this is a legacy script you must prepend the type (e.g. font, overlay).
    ```
* If source build, which git commit?
* System locale, i.e. the output of `echo "$LANG $LC_NUMERIC"`:
  Before reporting a rendering issue, try running RViz with `LANG=C rviz`!

rhaschke, 2022-02-21:
I assume that the problem doesn't occur with system Ogre 1.9? I'm afraid I won't have time to look into this issue this week.
By the way, is a newer version of Ogre released in Ubuntu 22?

simonschmeisser, 2022-02-21:
> I assume that the problem doesn't occur with system Ogre 1.9? I'm afraid I won't have time to look into this issue this week.
No, works fine with 1.9   Just thought it might ring a bell with you. I'll try to debug it further one of the evenings

> By the way, is a newer version of Ogre released in Ubuntu 22?
It will contain at least 1.12.10 and maybe 1.12.13 but that is progressing slowly and might not make the import deadline. 1.9 will still be around though

simonschmeisser, 2022-03-18:
Will be 1.12.10 (and 1.9 of course), I missed the deadline for going to 1.12.13