ros-visualization/rviz #1720: The topic in the added GridCells plug-in is not displayed

qisongH, 2022-03-07:
After adding the GridCells plug-in through the "Add - > By topic" interface in RViz, the topic in the added GridCells plug-in is not displayed, which is different from that of other plug-ins added in the same way.
[image]
[image]

### Your environment
* OS Version: e.g. Ubuntu 18.04
* ROS Distro: [Kinetic | Melodic | Noetic]
* RViz, Qt, OGRE, OpenGl version as printed by rviz:

[image]

rhaschke, 2022-03-07:
You mean, the topic field is empty, while it should show the selected topic name?
Which version of rviz are you using?

rhaschke, 2022-03-07:
I was able to reproduce the issue on Melodic, but not on Noetic. Can you confirm?

qisongH, 2022-03-07:
> You mean, the topic field is empty, while it should show the selected topic name? Which version of rviz are you using?
Yes, the topic field should show the selected topic name and display GridCells in 3D area. The version of rviz is 1.13.18.

qisongH, 2022-03-07:
> I was able to reproduce the issue on Melodic, but not on Noetic. Can you confirm? I won't dig into the code to fix a bug that's only present in an outdated release. Please consider updating or file a fixing PR for Melodic. Thanks.

Sorry, this issue occurred on Melodic, I re-edited the environment in which the issue occurred.

rhaschke, 2022-07-31:
Fixed via #1722

closed 2022-07-31 by rhaschke