ros-visualization/rviz #1727: Playing ROS bag which contains many tf data whole night make rviz ui stuck

tsukione, 2022-03-18:
The rosbag is too large. I can't upload it temporarily. But may be it easy to make a new one.

My bag only contains point_cloud2 and many tf data。
point_cloud2 frequency is 10hz
tf tree is just like this:
map
      body
             lidar_roof_node
             lidar_front_node
             lidar_side_node
             lidar_rear_node
tf frequency is 100hz

I tried to find out why. but I only find the call time of transformPose is growing. I still not figure out why the calling time of this line will growing.
[tf_->transformPose(fixed_frame_, pose_in, pose_out)](https://github.com/ros-visualization/rviz/blob/16df936f16ee7e1c50bf439c5798a1e8d40d842a/src/rviz/frame_manager.cpp#L276)

Do you have any idea? that will help me a lot, thanks @rhaschke 

### Your environment
* OS Version:  Ubuntu 18.04
* ROS Distro: [Melodic]
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
    ```
 [ INFO] [1647573397.962159400]: rviz version 1.13.23
[ INFO] [1647573397.962215100]: compiled against Qt version 5.9.5
[ INFO] [1647573397.962249800]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1647573397.972815000]: Forcing OpenGl version 0.
[ INFO] [1647573398.557273500]: Stereo is NOT SUPPORTED
[ INFO] [1647573398.557366000]: OpenGL device: llvmpipe (LLVM 10.0.0, 256 bits)
[ INFO] [1647573398.557440600]: OpenGl version: 3.1 (GLSL 1.4).

    ```
* If source build, which git commit?
* System locale, i.e. the output of `echo "$LANG $LC_NUMERIC"`:
  Before reporting a rendering issue, try running RViz with `LANG=C rviz`!

rhaschke, 2022-03-18:
About what time frame you are talking here? Does _running rviz the "whole night"_ means: it takes several hours before rviz slows down?
Can you reproduce the behavior with the test script [`send_lots_of_tf.py`](https://github.com/ros-visualization/rviz/blob/noetic-devel/src/test/send_lots_of_tf.py)

> I tried to find out why. but I only find the call time of transformPose is growing.

Generally, TF buffers past transforms. rviz does that for 10mins:
https://github.com/ros-visualization/rviz/blob/16df936f16ee7e1c50bf439c5798a1e8d40d842a/src/rviz/frame_manager.cpp#L58

Given 100Hz data, this can easily grow your memory. Did you check your memory consumption? Maybe the machine starts to swap?

tsukione, 2022-03-18:
Running the script [send_lots_of_tf.py](https://github.com/ros-visualization/rviz/blob/noetic-devel/src/test/send_lots_of_tf.py) won't reproduce the behavior. 
But, if I record the tf published from the script. then play the bag, the behavior reproduced. what happened if the timestamp is looping

rhaschke, 2022-03-18:
Please answer my remaining questions:
- How long do you need to record from [`send_lots_of_tf.py`](https://github.com/ros-visualization/rviz/blob/noetic-devel/src/test/send_lots_of_tf.py) to reproduce the behavior?
- Did you monitor memory usage? Does memory increase continuously (after 10mins)? Does the system start to swap?

> What happens if the timestamp is looping?

Time starts over and all displays are reset (same as pressing Reset button):
https://github.com/ros-visualization/rviz/blob/968a654f35cc6d199c1122554fcf2f12be07f554/src/rviz/visualization_manager.cpp#L421-L426

tsukione, 2022-03-18:
> How long do you need to record from [send_lots_of_tf.py](https://github.com/ros-visualization/rviz/blob/noetic-devel/src/test/send_lots_of_tf.py) to reproduce the behavior?

tsukione, 2022-03-18:
> How long do you need to record from [send_lots_of_tf.py](https://github.com/ros-visualization/rviz/blob/noetic-devel/src/test/send_lots_of_tf.py) to reproduce the behavior?
```
path:        lots_of_tf_26s.bag
version:     2.0
duration:    26.2s
start:       Mar 18 2022 20:02:19.23 (1647604939.23)
end:         Mar 18 2022 20:02:45.42 (1647604965.42)
size:        14.2 MB
messages:    95644
compression: none [18/18 chunks]
types:       tf2_msgs/TFMessage [94810edda583a504dfda3829e70d7eec]
topics:      /tf   95644 msgs    : tf2_msgs/TFMessage
```
 > Did you monitor memory usage? Does memory increase continuously (after 10mins)?
at start: 105.0MB
after 5min: 194.3MB
after 10min: 287.2MB
after 20min: 480.0MB
after 25min: 567.8MB

It seems memory increasing linearly from the start time. And after 25min RViz's fps starts to be unstable, every time rosbag play start over, fps will drop down to 20fps.

>  Does the system start to swap?
no, only memory.

> Time starts over and all displays are reset (same as pressing Reset button):
It seems rviz did not call resetTime if timestamp is looping, I need to click the reset button manually to restore rviz.

rhaschke, 2022-03-18:
> And after 25min RViz's fps starts to be unstable, every time rosbag play start over, fps will drop down to 20fps.

Do you mean, rviz' fps drops to 20fps _only in a short time window_ around the point when rosbag starts over? Afterwards it settles around 30fps (the default) again?

> It seems rviz did not call `resetTime` if timestamp is looping?

For me this works. Probably you missed to configure [`use_sim_time=true`](http://wiki.ros.org/Clock#Using_Simulation_Time_from_the_.2Fclock_Topic)? If that's missing, your ROS time is identical to wall time and continuously progresses. Not configuring sim time, I see these warnings in the console (in ROS Noetic, using TF2):
`[ WARN] ros.rosconsole_bridge.console_bridge: TF_REPEATED_DATA ignoring data with redundant timestamp for frame frame0_15 at time 1647609868.234283 according to authority unknown_publisher`

tsukione, 2022-03-19:
> Do you mean, rviz' fps drops to 20fps only in a short time window around the point when rosbag starts over? Afterwards it settles around 30fps (the default) again?

Yes, fps will drop down lasts for more than 10 ms, and the it settles around 30fps again. As the time goes on, the frame drop becomes more and more severe. If it lasts long enough, it gets stuck completely.

> Probably you missed to configure [use_sim_time=true](http://wiki.ros.org/Clock#Using_Simulation_Time_from_the_.2Fclock_Topic)?

Got it,  If I configure [use_sim_time=true](http://wiki.ros.org/Clock#Using_Simulation_Time_from_the_.2Fclock_Topic) and play bag with --clock, RViz does not have anomalies. that solved my problem. Thanks very much 😄

closed 2022-03-22 by tsukione