ros-visualization/rviz #1728: rviz will coredump after I delete all the buttons on the toolbar and then save it

solookin, 2022-03-18:
When I delete all the buttons on the toolbar by using "-" button, and then save it,  I found that rviz will coredump.

### Your environment
[ INFO] [1647590915.554002200]: rviz version 1.13.17
[ INFO] [1647590915.554056100]: compiled against Qt version 5.9.5
[ INFO] [1647590915.554064200]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1647590915.557514500]: Forcing OpenGl version 0.
[ INFO] [1647590915.655703900]: Stereo is NOT SUPPORTED
[ INFO] [1647590915.655789700]: OpenGL device: llvmpipe (LLVM 10.0.0, 256 bits)
[ INFO] [1647590915.655827700]: OpenGl version: 3.1 (GLSL 1.4).

rhaschke, 2022-03-18:
Thanks for reporting. Obviously, it doesn't make sense to remove all tools. Namely, you should keep `Interact` :wink:

solookin, 2022-03-19:
> 

OK, I got it

solookin, 2022-03-19:
> Thanks for reporting. Obviously, it doesn't make sense to remove all tools. Namely, you should keep `Interact` 😉

OK, I got it

closed 2022-03-31 by solookin

rhaschke, 2022-03-31:
Fixed via #1729.

PR #1729 (2022-03-18): Fix segfault when no tools are available
Fixes #1728.