ros-visualization/rviz #1738: Associated widget which is docked creating a tabbed panels gets disabled if tab is not visible

Levi-Armstrong, 2022-04-20:
Currently if I dock my panel on top of another panel creating tabbed panels and then you switched tab it disables the display. Is there a way to disable this functionality?

Levi-Armstrong, 2022-04-20:
I believe this a related [issue](https://github.com/ros-visualization/rviz/issues/940) but the issue shows resolved but this still appears to be happening on noetic version.

Levi-Armstrong, 2022-04-20:
Tab Selected
[image]

Tab Unselected
[image]

Levi-Armstrong, 2022-04-20:
Then I can enable the display from the tree and the scene is visible and the tab is not which is the behavior expected.
[image]

Levi-Armstrong, 2022-04-20:
A current work around is to add the following connection enabling the display on any visibility change to the panel.

``` c++
connect(getAssociatedWidgetPanel(), SIGNAL( visibilityChanged( bool ) ), this, SLOT( onAssociatedPanelVisibilityChange( bool ) ));
```

``` c++
void WorkbenchDisplay::onAssociatedPanelVisibilityChange(bool /*visible*/)
{
  setEnabled(true);
}
```

Levi-Armstrong, 2022-04-20:
Well actually that does not work because now the display cannot be disabled.

Levi-Armstrong, 2022-04-20:
Finally was able to create a work around.

I first had to disconnect the following signal.

``` c++
disconnect(getAssociatedWidgetPanel(), SIGNAL( visibilityChanged(bool) ), this, SLOT( associatedPanelVisibilityChange(bool) ));
```

Then override the `onEnableChanged` with the following
``` c++
void WorkbenchDisplay::onEnableChanged()
{
  QApplication::setOverrideCursor(QCursor(Qt::WaitCursor));
  queueRender();
  /* We get here, by two different routes:
   * - First, we might have disabled the display.
   *   In this case we want to close/hide the associated widget.
   *   But there is an exception: tabbed DockWidgets shouldn't be hidden, because then we would loose the
   * tab.
   * - Second, the corresponding widget changed visibility and we got here via
   * associatedPanelVisibilityChange().
   *   In this case, it's usually counterproductive to show/hide the widget here.
   *   Typical cases are: main window was minimized/unminimized, tab was switched.
   */
  if (isEnabled())
  {
    scene_node_->setVisible(true);

    if (getAssociatedWidgetPanel() != nullptr)
    {
      getAssociatedWidgetPanel()->show();
      getAssociatedWidgetPanel()->setEnabled(true);
    }
    else if (getAssociatedWidget() != nullptr)
    {
      getAssociatedWidget()->show();
      getAssociatedWidgetPanel()->setEnabled(true);
    }

    if (isEnabled()) // status might have changed, e.g. if show() failed
      onEnable();
  }
  else
  {
    onDisable();

    if (getAssociatedWidgetPanel() != nullptr)
      getAssociatedWidgetPanel()->setDisabled(true);
    else if (getAssociatedWidget())
      getAssociatedWidget()->setDisabled(true);

    scene_node_->setVisible(false);
  }
  QApplication::restoreOverrideCursor();
}
```
This just disables the widget opposed to hiding and showing.

rhaschke, 2022-04-21:
There is a simpler solution: #1739.

Levi-Armstrong, 2022-04-21:
> There is a simpler solution: #1739.

Thanks for the fix.

Levi-Armstrong, 2022-04-25:
Sure, it is under a lot of development at the moment. [tesseract_gui](https://github.com/tesseract-robotics/tesseract_gui)

PR #1739 (2022-05-13): Don't disable display if associated widget tab changes
Fixes #1738

If an associated widget is tabbed and deselected, its visibility changes to false, which triggers disabling the display.
However, in this case, the widget is still there and thus the display shouldn't be disabled.
closed 2022-05-13 by rhaschke