ros-visualization/rviz #1740: Set the type of toolbar button used for rviz::Tool plugin

Levi-Armstrong, 2022-04-25:
Is it possible to change the type of button used for a tool plugin?

rhaschke, 2022-04-26:
No. The tool is just represented by a `QAction` added to a [`QToolBar`](https://doc.qt.io/qt-5/qtoolbar.html), which always creates a `QToolButton`. The Qt API to add arbitrary widgets is not exposed. What type of button do you think of?

Levi-Armstrong, 2022-04-26:
I wanted to add add a drop down menu like the button to remove tools.

rhaschke, 2022-04-26:
> I wanted to add add a drop down menu like the button to remove tools.

Actually, the remove-tools thing isn't a drop down combobox, but a [menu of actions](https://github.com/ros-visualization/rviz/blob/1f622b8c95b8e188841b5505db2f97394d3e9c6c/src/rviz/visualization_frame.cpp#L527-L535).
How exactly should the selection in this list (menu or real drop down) effect the behavior of the tool? Are those entries considered "sub-tools" of your tool plugin that you would like to group?

Currently, for each added tool, this slot function simply adds a corresponding action (and tool button):
https://github.com/ros-visualization/rviz/blob/1f622b8c95b8e188841b5505db2f97394d3e9c6c/src/rviz/visualization_frame.cpp#L1167-L1180

Levi-Armstrong, 2022-04-26:
I wanted to add a tool for changing the rviz theme where you could select (Dark, Light and Default). Right now the tool adds an enum property to the tools property but it would be nice if you could do this form the toolbar with a drop-down menu. On selection it would set the theme.

rhaschke, 2022-04-26:
I see. I don't think that should be done with the tool plugin. A tool primarily defines mouse and keyboard interaction.
However, another suitable plugin class doesn't exist either. You could try dynamic casting `context_->getWindowManager()` to `QMainWindow` and then access its toolbar via: `findChild<QToolBar*>("Tools")`.

Levi-Armstrong, 2022-04-26:
I will give that a try and report report back.

Levi-Armstrong, 2022-04-26:
I assume for this to work it would still need to be associated with one of the available plugin types?

rhaschke, 2022-04-26:
> I assume for this to work it would still need to be associated with one of the available plugin types?

Sure.

Levi-Armstrong, 2022-05-13:
@rhaschke  The propose solution worked. Thank you for the help.

closed 2022-05-13 by Levi-Armstrong