ros-visualization/rviz #1748: Invisible robot model meshes (in WSL?)

jorolf, 2022-05-23:
The `RobotModel` is invisible even though there are no visible errors:
[image]
Adding a `TF` display shows the frames of the robot correctly:
[image]
I tried both the panda and the fanuc robot and neither displayed the robot mesh

### Your environment
* OS Version: Windows 11 Pro 21H2 (22000.675) running WSL2 with Ubuntu 20.04
* ROS Distro: Noetic
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
    ```
    rviz version 1.14.14
    compiled against Qt version 5.12.8
    compiled against OGRE version 1.9.0 (Ghadamon)
    Forcing OpenGl version 0.
    Stereo is NOT SUPPORTED
    OpenGL device: D3D12 (NVIDIA GeForce GTX 1050)
    OpenGl version: 3.1 (GLSL 1.4).
    ```
* `echo "$LANG, $LC_NUMERIC"` prints `C.UTF-8`
    Using `LANG=C rviz` does not help
* Running `glxinfo` prints this:
  <details>
    <summary>Click to expand</summary>
    
    ```
    direct rendering: Yes
    server glx vendor string: SGI
    server glx version string: 1.4
    server glx extensions:
        GLX_ARB_context_flush_control, GLX_ARB_create_context,
        GLX_ARB_create_context_no_error, GLX_ARB_create_context_profile,
        GLX_ARB_fbconfig_float, GLX_ARB_framebuffer_sRGB, GLX_ARB_multisample,
        GLX_EXT_create_context_es2_profile, GLX_EXT_create_context_es_profile,
        GLX_EXT_fbconfig_packed_float, GLX_EXT_framebuffer_sRGB,
        GLX_EXT_import_context, GLX_EXT_libglvnd, GLX_EXT_no_config_context,
        GLX_EXT_texture_from_pixmap, GLX_EXT_visual_info, GLX_EXT_visual_rating,
        GLX_MESA_copy_sub_buffer, GLX_OML_swap_method, GLX_SGIS_multisample,
        GLX_SGIX_fbconfig, GLX_SGIX_pbuffer, GLX_SGIX_visual_select_group,
        GLX_SGI_make_current_read
    client glx vendor string: Mesa Project and SGI
    client glx version string: 1.4
    client glx extensions:
        GLX_ARB_context_flush_control, GLX_ARB_create_context,
        GLX_ARB_create_context_no_error, GLX_ARB_create_context_profile,
        GLX_ARB_create_context_robustness, GLX_ARB_fbconfig_float,
        GLX_ARB_framebuffer_sRGB, GLX_ARB_get_proc_address, GLX_ARB_multisample,
        GLX_ATI_pixel_format_float, GLX_EXT_buffer_age,
        GLX_EXT_create_context_es2_profile, GLX_EXT_create_context_es_profile,
        GLX_EXT_fbconfig_packed_float, GLX_EXT_framebuffer_sRGB,
        GLX_EXT_import_context, GLX_EXT_no_config_context, GLX_EXT_swap_control,
        GLX_EXT_swap_control_tear, GLX_EXT_texture_from_pixmap,
        GLX_EXT_visual_info, GLX_EXT_visual_rating, GLX_INTEL_swap_event,
        GLX_MESA_copy_sub_buffer, GLX_MESA_multithread_makecurrent,
        GLX_MESA_query_renderer, GLX_MESA_swap_control, GLX_NV_float_buffer,
        GLX_OML_swap_method, GLX_OML_sync_control, GLX_SGIS_multisample,
        GLX_SGIX_fbconfig, GLX_SGIX_pbuffer, GLX_SGIX_visual_select_group,
        GLX_SGI_make_current_read, GLX_SGI_swap_control, GLX_SGI_video_sync
    GLX version: 1.4
    GLX extensions:
        GLX_ARB_context_flush_control, GLX_ARB_create_context,
        GLX_ARB_create_context_no_error, GLX_ARB_create_context_profile,
        GLX_ARB_fbconfig_float, GLX_ARB_framebuffer_sRGB,
        GLX_ARB_get_proc_address, GLX_ARB_multisample,
        GLX_EXT_create_context_es2_profile, GLX_EXT_create_context_es_profile,
        GLX_EXT_fbconfig_packed_float, GLX_EXT_framebuffer_sRGB,
        GLX_EXT_import_context, GLX_EXT_no_config_context,
        GLX_EXT_texture_from_pixmap, GLX_EXT_visual_info, GLX_EXT_visual_rating,
        GLX_MESA_copy_sub_buffer, GLX_MESA_query_renderer, GLX_OML_swap_method,
        GLX_SGIS_multisample, GLX_SGIX_fbconfig, GLX_SGIX_pbuffer,
[... 215 lines cut ...]
        GL_OES_vertex_half_float
    ```
  </details>

### How to reproduce
Setup Ubuntu VM:
Inside Powershell or Cmd run
```
wsl --install -d Ubuntu-20.04
```
and follow the instructions.

After installation run `sudo apt update` and `sudo apt upgrade`.
Then follow [this tutorial](http://wiki.ros.org/noetic/Installation/Ubuntu) until step 1.4 where `ros-noetic-desktop-full` should be installed.
Finally install `ros-noetic-moveit` and `ros-noetic-moveit-resources` and run:
```
roslaunch moveit_resources_panda_moveit_config test_environment.launch &
rviz
```

It used to work a few weeks ago, I'm not exactly sure what broke it. (Maybe `apt upgrade`?) 
It could be something related to the shader considering that it's a rendering only bug?
I'm also not sure if this is a local problem since I only have a single Windows 11 machine to test this on. (which supports WSLg)

If there's anything I have missed or I should try let me know.

rhaschke, 2022-05-23:
I can confirm this issue on another WSLg installation, also employing an NVIDIA card.
I filed an issue upstream: https://github.com/microsoft/wslg/issues/744

rhaschke, 2022-05-26:
A laptop with an Intel GPU only, doesn't exhibit this issue. It seems to be related to Microsoft's D3D12 renderer.

rhaschke, 2022-05-26:
As a **workaround**, you can set the environment variable `LIBGL_ALWAYS_SOFTWARE=true`, thus disabling accelerated rendering and instead enabling software rendering via `llvmpipe`.