ros-visualization/rviz #1753: PointCloud crashes when switching .rviz files

ysl-design, 2022-06-26:
Describe your issue here and explain how to reproduce it.

The description may be a little too much, please be patient to read it : )

### Your environment
* OS Version: Ubuntu 18.04
* ROS Distro: Melodic
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
    ```
  [ INFO] [1656234735.426802600]: rviz version 1.13.24
  [ INFO] [1656234735.426857300]: compiled against Qt version 5.9.5
  [ INFO] [1656234735.426868900]: compiled against OGRE version 1.9.0 (Ghadamon)
  [ INFO] [1656234735.438341600]: Forcing OpenGl version 0.
  [ INFO] [1656234735.741489900]: Stereo is NOT SUPPORTED
  [ INFO] [1656234735.741942600]: OpenGL device: llvmpipe (LLVM 10.0.0, 256 bits)
  [ INFO] [1656234735.741979000]: OpenGl version: 3.1 (GLSL 1.4).
    ```
* build against tag 1.13.24
* System locale,:  zh_CN.UTF-8

My scenario:
Hi, I've added a dozen display plugins to rviz, including pointcloud2, marker, markerArray, etc. I then saved the settings to the xxx.rviz file. Later, I loaded the xxx.rviz file several times and opened the file through ‘File -> Open Config’. (The corresponding data is still being sent when the config file is switched.) I found that rviz occasionally crashed.

I found that there are three reasons for crashing, all related to the PointCloudCommon class:
(1) Based on the backtrace and code analysis, it is found that an emitTimeSignal signal is sent in the PointCloudCommon::processMessage function. This signal transfers the pointer pointing to the pointcloud2 plug-in to the TimePanel::onTimeSignal function. In some cases, the pointcloud2 plug-in is destroyed before TimePanel::onTimeSignal is executed. As a result, the display pointer transferred to the TimePanel::onTimeSignal function becomes invalid, and a segment fault occurs when an invalid memory is accessed. 
I wonder if you can avoid this by adding a judgment at the top of the TimePanel::onTimeSignal function that determines whether sender() is a null pointer.

the backtrace shows that:
```
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Core was generated by `rviz -d /home/ysl/Download/critical/RViz'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x00007f1403a289b7 in rviz::TimePanel::onTimeSignal(rviz::Display*, ros::Time) () from /opt/ros/melodic/lib/librviz.so
[Current thread is 1 (Thread 0x7f1403e94cc0 (LWP 4579))]
(gdb) bt
#0  0x00007f1403a289b7 in rviz::TimePanel::onTimeSignal(rviz::Display*, ros::Time) () at /opt/ros/melodic/lib/librviz.so
#1  0x00007f1403a71a33 in  () at /opt/ros/melodic/lib/librviz.so
#2  0x00007f1402bc5092 in QObject::event(QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#3  0x00007f14031e974b in QWidget::event(QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#4  0x00007f14031aa83c in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#5  0x00007f14031b2104 in QApplication::notify(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#6  0x00007f1402b958a8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#7  0x00007f1402b9801d in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#8  0x00007f1402bef233 in  () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#9  0x00007f13fbf2d537 in g_main_context_dispatch () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#10 0x00007f13fbf2d770 in  () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#11 0x00007f13fbf2d7fc in g_main_context_iteration () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#12 0x00007f1402bee85f in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#13 0x00007f1402bc4525 in QMetaObject::activate(QObject*, int, int, void**) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#14 0x00007f1403a6ff02 in rviz::VisualizationFrame::statusUpdate(QString const&) () at /opt/ros/melodic/lib/librviz.so
#15 0x00007f1402bc4525 in QMetaObject::activate(QObject*, int, int, void**) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#16 0x00007f1403a70035 in rviz::VisualizationManager::statusUpdate(QString const&) () at /opt/ros/melodic/lib/librviz.so
#17 0x00007f1403a56ca1 in rviz::VisualizationManager::load(rviz::Config const&) () at /opt/ros/melodic/lib/librviz.so
#18 0x00007f1403a5226b in rviz::VisualizationFrame::load(rviz::Config const&) () at /opt/ros/melodic/lib/librviz.so
#19 0x00007f1403a53091 in rviz::VisualizationFrame::loadDisplayConfigHelper(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) () at /opt/ros/melodic/lib/librviz.so
#20 0x00007f1403a5325c in rviz::VisualizationFrame::loadDisplayConfig(QString const&) () at /opt/ros/melodic/lib/librviz.so
#21 0x00007f1403a540d9 in rviz::VisualizationFrame::onOpen() () at /opt/ros/melodic/lib/librviz.so
#22 0x00007f1402bc4525 in QMetaObject::activate(QObject*, int, int, void**) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#23 0x00007f14031a4122 in QAction::triggered(bool) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#24 0x00007f14031a680c in QAction::activate(QAction::ActionEvent) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#25 0x00007f140332305c in  () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#26 0x00007f140332a50b in  () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#27 0x00007f140332b333 in QMenu::mouseReleaseEvent(QMouseEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#28 0x00007f14031e9038 in QWidget::event(QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#29 0x00007f140332d65b in QMenu::event(QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#30 0x00007f14031aa83c in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#31 0x00007f14031b265f in QApplication::notify(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#32 0x00007f1402b958a8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#33 0x00007f14031b1632 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
[... 116 lines cut ...]
#15 0x00007f49040c567d in rviz::VisualizationFrame::onOpen() (this=0x55e6dfa6bc00) at /home/Downloads/Downloads/rviz/rviz/src/rviz/visualization_frame.cpp:1090
#16 0x00007f4903f8fab6 in rviz::VisualizationFrame::qt_static_metacall(QObject*, QMetaObject::Call, int, void**) (_o=0x55e6dfa6bc00, _c=QMetaObject::InvokeMetaMethod, _id=7, _a=0x7ffd5b9d2ff0)
    at /home/Downloads/Downloads/rviz/rviz/cmake-build-debug/src/rviz/rviz_autogen/EWIEGA46WW/moc_visualization_frame.cpp:224
#17 0x00007f4903010525 in QMetaObject::activate(QObject*, int, int, void**) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#18 0x00007f49035f0122 in QAction::triggered(bool) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#19 0x00007f49035f280c in QAction::activate(QAction::ActionEvent) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#20 0x00007f49035f30d5 in QAction::event(QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#21 0x00007f49035f683c in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#22 0x00007f49035fe104 in QApplication::notify(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#23 0x00007f4902fe18a8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#24 0x00007f48fd99f4b7 in QShortcutMap::dispatchEvent(QKeyEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
#25 0x00007f48fd99f58a in QShortcutMap::tryShortcut(QKeyEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
#26 0x00007f48fd94dcb3 in QWindowSystemInterface::handleShortcutEvent(QWindow*, unsigned long, int, QFlags<qt::KeyboardModifier>, unsigned int, unsigned int, unsigned int, QString const&, bool, unsigned short) () at /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
#27 0x00007f48fd96e047 in QGuiApplicationPrivate::processKeyEvent(QWindowSystemInterfacePrivate::KeyEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
#28 0x00007f48fd973095 in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
[... cut ...]
#33 0x00007f48fc37c7fc in g_main_context_iteration () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#34 0x00007f490303a85f in QEventDispatcherGlib::processEvents(QFlags<qeventloop::ProcessEventsFlag>) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#35 0x00007f4902fdf8da in QEventLoop::exec(QFlags<qeventloop::ProcessEventsFlag>) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#36 0x00007f4902fe8984 in QCoreApplication::exec() () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#37 0x000055e6de4bedcf in main ()
```

rhaschke, 2022-06-27:
Thanks for this detailed error report.
I think the clean solution would be to stop calling `processMessage` / waiting for a running call to finish _before_ destructing the point cloud.

ysl-design, 2022-06-28:
Thank you for your response and for the corrections to the issues in my description.
I tested it with the modifications in PR #1754  to compile RViz, and the problem (1) in my description has been fixed. However, problems (2) and (3) in the description still exist during the test.
I think that even though it stops subscribing to new data before destruction, threads that have entered processMessage will continue after the PointCloudCommon destruction is complete, and I suspect this may be the cause of the crash during mutex destruction.

rhaschke, 2022-06-28:
> I think that even though it stops subscribing to new data before destruction, threads that have entered processMessage will continue after the PointCloudCommon destruction is complete, and I suspect this may be the cause of the crash during mutex destruction.

Yeah, that might be an issue. Quoting the doc: Attempting to destroy a locked mutex results in undefined behavior.
I have pushed another commit to ensure the mutexes are held by the destructor...

ysl-design, 2022-06-29:
Thank you for your reply and modification. I'm sorry I didn't get back to you in time.
I tried new modifications and it looks like the problem (3) in the description has been fixed as well. But problem (2) still exists.
According to the analysis of the core dump file again, problem (2) occurs in this case: After PointCloud2Display is destructed, its parent class MessageFilterDisplay is destroyed. A crash occurred while executing delete tf_filter_ in the ~MessageFilterDisplay() function.
The backtrace information ultimately points to the Signal1 class in the message_filters of the ROS. The mutex_ in the Signal1 class was destroyed without being unlocked, resulting in a crash.
Can this problem be avoided when MessageFilterDisplay is destructed or the code related to ROS needs to be modified?

rhaschke, 2022-06-30:
The memory leaks reported are not related to your issue. If you want to report some, only consider those related to rviz. There are many low-level libraries having leaks, which we cannot fix anyway.

That you can't reproduce the issue (2) anymore might be related to the slower execution with `asan`.

I don't understand why the `mutex_` in `MessageFilter/Signal1` can still become locked after disconnecting the `message_connection_` (which should stop pushing new messages) and clearing the message buffer. I was hoping for more insight with asan...
Could you build with `-DCMAKE_BUILD_TYPE=RelWithDebInfo` again and check which thread is holding the lock at crash time? Please paste the backtrace of this thread (which should include `message_filters::Signal1::call()`).

ysl-design, 2022-07-01:
I used -DCMAKE_BUILD_TYPE=RelWithDebInfo to build rviz, and again, it no longer crashes.
I remove these compilation parameters and compile and run rviz, and it crashes again. I added print statements before and after mutex_ is locked and unlocked in the Signal1 class. According to the print information, after mutex_ is locked in the `void Signal1::call(const ros::MessageEvent<M const>& event)` function of signal1, mutex_ is not unlocked until the rviz crashes. According to the core dump file, the crash occurred during the destruction of the locked mutex_.
The following is the `cout` result. At the end of the `Signal1::call` function, the `cout` statement "call unlock" is not printed, indicating that mutex_ is not unlocked.
```
call lock 0x55ce07f77a68
...
~PointCloudCommon()
~PointCloud2Display() 0x55ce0754d100
~MessageFilter() start
removeCallback lock 0x55ce0754d330
removeCallback unlock 0x55ce0754d330
~MessageFilter() end
rviz: /usr/include/boost/thread/pthread/mutex.hpp:111：boost::mutex::~mutex(): Assertion '!res' failed
Aborted (core dumped)
```
According to other printed information, mutex_ is locked in the `Signal1::call` function, then, the program runs to `helper->call(event, nonconst_force_copy)` and crashes.

According to the core dump file, the address of the object when the crash occurs is **0x55ce07f77a68**, which is the same as the address of the object that invokes the `Signal1::call` function in the `cout` information. 
```
...
#4  0x00007fb805091160 in boost::mutex::~mutex() (this=0x55ce07f77a68, __in_chrg=<optimized out>) at /usr/include/boost/thread/pthread/mutex.hpp:111
...
```
This should indicate that the crash occurred in the Signal1::call function and that mutex_ was not unlocked.

rhaschke, 2022-07-01:
Thanks for your investigation. I continued as well and traced the issue down to `tf2_ros::MessageFilter`. I filed an initial PR
https://github.com/ros/geometry2/pull/538, but there are more deadlocks to resolve...

rhaschke, 2022-07-01:
> I used `-DCMAKE_BUILD_TYPE=RelWithDebInfo` to build rviz, and again, it no longer crashes.

A release build disables all assertions. Hence, it is not aborting anymore (due to failing assertions).

rhaschke, 2022-07-27:
Fixed via #1754

closed 2022-07-27 by rhaschke