ros-visualization/rviz #1757: map shows incorrectly with wsl2 with rt6600

One2INF, 2022-07-24:
I follow [this](https://github.com/ishkapoor2000/Install_ROS_Noetic_On_WSL) article to setup my wsl2+ros environment.
But the map shows incorrectly. It seems only half of map shows.
[image]

* The same package works well in vmware+ubuntu.
* I test it with another computer(AMD Ryzen 7 5700G with Core Graphics card). The problem happen again.

### Your environment
* OS Version: Windows 11 Pro 21H2 (22000.832) running WSL2 with Ubuntu 20.04
* ROS Distro: Noetic
* RViz, Qt, OGRE, OpenGl version as printed by rviz.
```
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-shizj'
[ INFO] [1658671854.148744415]: rviz version 1.14.14
[ INFO] [1658671854.148790281]: compiled against Qt version 5.12.8
[ INFO] [1658671854.148808815]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1658671854.156160122]: Forcing OpenGl version 0.
[ INFO] [1658671855.144689454, 876.805000000]: Stereo is NOT SUPPORTED
[ INFO] [1658671855.144768672, 876.805000000]: OpenGL device: D3D12 (AMD Radeon RX 6600)
[ INFO] [1658671855.144823705, 876.805000000]: OpenGl version: 4.2 (GLSL 4.2) limited to GLSL 1.4 on Mesa system.
```
* System locale
echo "$LANG, $LC_NUMERIC" prints C.UTF-8
* Graphics card:
```
direct rendering: Yes
server glx vendor string: SGI
server glx version string: 1.4
server glx extensions:
client glx vendor string: Mesa Project and SGI
client glx version string: 1.4
client glx extensions:
GLX version: 1.4
GLX extensions:
Extended renderer info (GLX_MESA_query_renderer):
    Vendor: Microsoft Corporation (0xffffffff)
    Device: D3D12 (AMD Radeon RX 6600) (0xffffffff)
    Version: 22.2.0
    Accelerated: yes
    Video memory: 16291MB
    Unified memory: no
    Preferred profile: core (0x1)
    Max core profile version: 4.2
    Max compat profile version: 4.2
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.1
```

One2INF, 2022-07-25:
I solved it by ```export LIBGL_ALWAYS_SOFTWARE=true``` as mentioned in  ##1748.
I will verify it in another computer.

closed 2022-07-25 by One2INF

rhaschke, 2022-07-27:
Good that the work-around using software rendering helped you. However, please also upvote (:+1:) the upstream issue https://github.com/microsoft/wslg/issues/744. Thanks.