ros-visualization/rviz #1758: The plug-in in the display panel sometimes does not shrink as the display panel compresses.

ysl-design, 2022-07-27:
Describe your issue here and explain how to reproduce it.

### Your environment
* OS Version: Ubuntu 20.04
* ROS Distro: Noetic
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
    ```
   [ INFO] [1658890616.733419800]: rviz version 1.14.14
   [ INFO] [1658890616.733471700]: compiled against Qt version 5.12.8
   [ INFO] [1658890616.733501700]: compiled against OGRE version 1.9.0 (Ghadamon)
   [ INFO] [1658890616.741053500]: Forcing OpenGl version 0.
   [ INFO] [1658890616.894986100]: Stereo is NOT SUPPORTED
   [ INFO] [1658890616.895090400]: OpenGL device: llvmpipe (LLVM 12.0.0, 256 bits)
   [ INFO] [1658890616.895148600]: OpenGl version: 3.1 (GLSL 1.4).
    ```
* System locale, zh_CN.UTF-8

When I stretch the displays panel first, then compress it, and then double-click the plug-in component in the displays panel (such as the drop-down list), the picture below appears. It seems that the components don't adapt to size as the displays panel compresses. I do not know whether the problem is caused by the QT or the displays panel.
[image]

rhaschke, 2022-07-27:
I cannot reproduce this issue. Could you please provide more detailed instructions and a video as well?

https://user-images.githubusercontent.com/5376030/181205794-3ee24b7f-cc94-46ae-82f4-a135c89b42a6.mp4

ysl-design, 2022-07-27:
Sorry, I can't upload the video right now. The procedure is as follows:
### step1:
[image]
### step2:
[image]
### step3:
[image]
### result:
[image]

rhaschke, 2022-07-29:
Thanks for the clarification. Now, I can reproduce the issue.

rhaschke, 2022-07-31:
Could you please test https://github.com/ros-visualization/rviz/pull/1760?

ysl-design, 2022-08-01:
I tried #1760  and the problem did not recur.

rhaschke, 2022-08-01:
Closed via #1760.

PR #1760 (2022-08-01): Improve SplitterHandle of PropertyTreeWidgets
Various fixes and improvements to the SplitterHandle used in PropertyTreeWidgets. Fixes #1758.

| before | after |
|----------|---------|
| [image] | [image] |
closed 2022-08-01 by rhaschke