ros-visualization/rviz #1762: The Image plug-in displays abnormal images in certain sizes

qisongH, 2022-08-04:
When dragging the Image plug-in, **a row or column of abnormal pixels sometimes appear at the boundary of the image**. The following figure shows the exception.

[image]
The size of the Image plug-in is about *240×217*, and the **lower boundary pixels** are displayed incorrectly.

[image]
The size of the Image plug-in is about *433×155*, and the **right boundary pixels** are displayed incorrectly.

### Your environment
* OS Version: e.g. Ubuntu 18.04
* ROS Distro: [Melodic]
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
```
 [ INFO] [1646653665.369882000]: rviz version 1.13.18
 [ INFO] [1646653665.369933300]: compiled against Qt version 5.9.5
 [ INFO] [1646653665.369944800]: compiled against OGRE version 1.9.0 (Ghadamon)
 [ INFO] [1646653665.373622300]: Forcing OpenGl version 0.
 [ INFO] [1646653665.491788300]: Stereo is NOT SUPPORTED
 [ INFO] [1646653665.492296300]: OpenGL device: llvmpipe (LLVM 10.0.0, 256 bits)
 [ INFO] [1646653665.492382900]: OpenGl version: 3.1 (GLSL 1.4).
```

* OS Version: e.g. Ubuntu 20.04
* ROS Distro: [Noetic]
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
```
[ INFO] [1659601560.538959700]: rviz version 1.14.14
[ INFO] [1659601560.539014300]: compiled against Qt version 5.12.8
[ INFO] [1659601560.539023500]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1659601560.547179000]: Forcing OpenGl version 0.
[ INFO] [1659601560.706482400]: Stereo is NOT SUPPORTED
[ INFO] [1659601560.706561400]: OpenGL device: llvmpipe (LLVM 12.0.0, 256 bits)
[ INFO] [1659601560.706581800]: OpenGl version: 3.1 (GLSL 1.4).
```

rhaschke, 2022-08-05:
Thanks for reporting. Could you please review #1763?

qisongH, 2022-08-08:
> Thanks for reporting. Could you please review #1763?

Thanks for the fix.

rhaschke, 2023-03-29:
Closed via #1763.

PR #1763 (2022-08-08): Fix image display wrapping
Fixes #1762 resulting from texture wrapping. Switching to clamping fixes the problem.
closed 2023-03-29 by rhaschke