ros-visualization/rviz #1764: After the display plug-in is unchecked, objects displayed in the 3D area still exist.

ysl-design, 2022-08-08:
Describe your issue here and explain how to reproduce it.

My scenario goes like this: I construct a PointStamped data and send it at 30 Hz to the rviz for display. When the PointStamped plug-in is checked and unchecked several times, the following phenomenon occurs occasionally: The plug-in is unchecked, but objects are still displayed in the 3D area. This phenomenon seems to occur only in the noetic version, as I cannot reproduce it in the melodic.
[image]

The PointStamped data I constructed is as follows:
header: 
  seq: 341
  stamp: 
    secs: 1590459933
    nsecs: 777035282
  frame_id: "map"
point: 
  x: 1.2
  y: 2.2
  z: 3.2

### Your environment
* OS Version: Ubuntu 20.04
* ROS Distro: Noetic
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
    ```
    [ INFO] [1659950216.627252300]: rviz version 1.14.14
    [ INFO] [1659950216.627324600]: compiled against Qt version 5.12.8
    [ INFO] [1659950216.627357300]: compiled against OGRE version 1.9.0 (Ghadamon)
    [ INFO] [1659950216.637553100]: Forcing OpenGl version 0.
    [ INFO] [1659950216.798892800]: Stereo is NOT SUPPORTED
    [ INFO] [1659950216.798973800]: OpenGL device: llvmpipe (LLVM 12.0.0, 256 bits)
    [ INFO] [1659950216.799027900]: OpenGl version: 3.1 (GLSL 1.4).
    ```

* System locale, i.e. the output of `echo "$LANG $LC_NUMERIC"`: zh_CN.UTF-8

rhaschke, 2022-08-08:
It might be possible that there was a message pending in the ROS callback queue, when you disabled the display.
This message would be processed after the display was disabled and thus show anyway. This should occur very rarely only, right?
I couldn't reproduce the issue on Noetic.

rhaschke, 2022-08-08:
Could you please verify whether #1765 fixes your issue?

ysl-design, 2022-08-08:
There is a low probability that this problem occurs. It occurs 2 to 3 times every 100 times checked and unchecked operation.
I tried this  #1765  and it did work, the problem did not recur.
I think not only PointStamped but also other plug-ins have the same problem. So I think this modification #1765 should be placed in the base class MessageFilterDisplay, perhaps the processTypeErasedMessage function?

rhaschke, 2022-08-08:
You are right. I shifted the check to the base class.

ysl-design, 2022-08-09:
I tried new modifications #1765 and found the problem reappeared. It seems that putting modifications in the incomingMessage function does not work. I tried putting the modifications in the processTypeErasedMessage function and the problem was fixed.

rhaschke, 2022-08-09:
Thanks for this hint. I merged a different version to Noetic.

PR #1765 (2022-08-08): PointStampedDisplay: Ignore incoming messages if disabled
Fixes #1764
closed 2022-08-09 by rhaschke