ros-visualization/rviz #1767: How do I switch the Linux graphics stack from GLX to EGL or how do I use RVIz under EGL?

cljidkx, 2022-08-15:
I can't run RVIz in arm64.Debugging problems through GDB occurs in glXCreateContext().
The glXCreateContext() is on line 154 of src/rviz/ogre_helpers/render_system.cpp.
```
(gdb) p attribList 
$2 = {4, 5, 12, 16, 13, 8, 0}
(gdb) p display
$1 = (Display *) 0x5555b436c0
(gdb) p visual
$4 = (XVisualInfo *) 0x0
(gdb) p dummy_window_id_ 
$5 = 6291457
(gdb) n
154	  GLXContext context = glXCreateContext(display, visual, nullptr, 1);
(gdb) 

Thread 1 "rviz" received signal SIGSEGV, Segmentation fault.
0x0000007ff6459448 in glXCreateContext () from /usr/lib/aarch64-linux-gnu/libGLX.so.0

```

### Environment
* OS Version: e.g. Ubuntu 20.04
* ROS Distro: Noetic
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
    ```
    Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
    [ INFO] [1660098735.437995898]: rviz version 1.14.14
    [ INFO] [1660098735.438106153]: compiled against Qt version 5.12.8
    [ INFO] [1660098735.438121612]: compiled against OGRE version 1.9.0 (Ghadamon)
    [ INFO] [1660098735.451337275]: Forcing OpenGl version 0.
    Using Wayland-EGL
    arm_release_ver of this libmali is 'g6p0-01eac0', rk_so_ver is '5'.
    Using the 'xdg-shell' shell integration
    Segmentation fault (core dumped)
    ```
* If source build, which git commit?
95189c16e06b7a6d353b71b8c9fcc6c7696c683a

rhaschke, 2022-08-15:
Looks like the creation of the `XVisualInfo` failed. rviz is written for GLX. Can you run other glx programs, e.g. `glxgears`?

cljidkx, 2022-08-15:
Unable to run glx programs.Because the system uses EGL, it cannot run GLX programs. EGL is an extension of OpenGL on the arm platform for the x window system, and its function is equivalent to the glx library under x86.
I find it depends on wayland because I can run rviz properly in X11 but not in wayland.
The chip is arm64 and both are EGL.

rhaschke, 2022-08-15:
Do you have XWayland installed?
https://stackoverflow.com/questions/47788578/how-come-glx-based-applications-can-be-run-on-wayland-on-ubuntu

cljidkx, 2022-08-16:
> 你有安装 XWayland 吗？ https://stackoverflow.com/questions/47788578/how-come-glx-based-applications-can-be-run-on-wayland-on-ubuntu

Yes, I installed xwayland, but I still can't run the GLX application. Are there any other dependencies missing from my system?

rhaschke, 2022-08-16:
I don't know your system. Please use the rich resources of the internet to resolve your basic GLX problem first.
This is an issue tracker for rviz related problems only.

closed 2022-08-16 by rhaschke