ros-visualization/rviz #1770: I want to get the key release event in rviz::Tool class

lidongxu111, 2022-10-17:
I want to get the key release event in rviz::Tool class

but the function processKeyEvent can only get pressed key event.

Is there any way to get released key event?

rhaschke, 2022-10-17:
Not yet. You would need to augment the class and build from source. Is there a special reason you need this event?

lidongxu111, 2022-10-17:
I've rewritten the ProcessKeyEvent function, but it can only receive keyboard press events, not keyboard release events

lidongxu111, 2022-10-17:
[image]

lidongxu111, 2022-10-17:
so，Why is this KeyRelease not responding

rhaschke, 2022-10-18:
Because RenderPanel is only overriding `keyPressEvent()` not `keyReleaseEvent()`:
https://github.com/ros-visualization/rviz/blob/6a234b4404c5ee71051f0b338ab8d5d2d931c27b/src/rviz/render_panel.h#L145

Again my question: Why do you need the release event? What do you want to achieve eventually?

lidongxu111, 2022-10-19:
If I want this condition to be responsive, what do I do：
else if (event->type() == QEvent::KeyRelease && event->key() == 16777248) {
shift_pressing_ = false;
}

rhaschke, 2022-10-19:
Indeed, if you need to receive KeyRelease events, you need to modify rviz itself as suggested in https://github.com/ros-visualization/rviz/issues/1770#issuecomment-1282533322

lidongxu111, 2022-10-19:
But how do I modify rviz？
Modify render_panel.h?Do I put a KeyReleaseEvent function in there？

rhaschke, 2022-10-19:
Yes, both in render_panel.h and render_panel.cpp. You need to include rviz in your catkin workspace of course.

lidongxu111, 2022-10-19:
I've changed  render_panel.h in a file on my own virtual machine，and I don't have this render_panel.cpp，But processKeyEvent doesn't work

lidongxu111, 2022-10-19:
[image]

rhaschke, 2022-10-19:
> I've changed render_panel.h in a file on my own virtual machine，and I don't have this render_panel.cpp

Obviously, you need to clone the rviz repo (this one) into your catkin workspace. Just changing render_panel.h in /opt/ros doesn't help at all.

lidongxu111, 2022-10-19:
omg，Thank you very much for your help.

closed 2023-03-29 by rhaschke