ros-visualization/rviz #1775: Rviz is unable to find the correct urdfdom version

sspatank, 2022-12-05:
When I try to run rviz through my commandline, it starts up without any of the plugins  and throws the following error. 

```
[ERROR] [1670253530.576930800]: PluginlibFactory: The plugin for class 'rviz/Orbit' failed to load.  
Error: Failed to load library /home/[my_username]/anaconda3/envs/ros_env/lib/librviz_default_plugin.so. 
Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. 
Error string: Could not load library (Poco exception = liburdfdom_world.so.1.0: cannot open shared object file: No such file or directory)
```

It looks like the plugin library `lib/librviz_default_plugin.so` is unable to find `liburdfdom_world.so.1.0`. 

When I ran `ldd lib/librviz_default_plugin.so | grep 'not found'` I got the following output:
```
        liburdfdom_world.so.1.0 => not found
        liburdfdom_model.so.1.0 => not found
```

When I searched for those libraries using `ls lib | grep urdfdom` I got a different version installed on my computer:
```
liburdfdom_model.so
liburdfdom_model.so.3.0
liburdfdom_model_state.so
liburdfdom_model_state.so.3.0
liburdfdom_sensor.so
liburdfdom_sensor.so.3.0
liburdfdom_world.so
liburdfdom_world.so.3.0
urdfdom
urdfdom_headers
```

Looking at the version installed with conda using `conda list | grep urdfdom` I get the following output:

```
urdfdom                   3.1.0                h068d047_1    conda-forge
urdfdom_headers           1.1.0                h924138e_0    conda-forge
```

It looks like the version installed is higher than the version librviz_default_plugin is looking for. Why is it looking for that version? Can I make it point to the higher version I have available? 

### Your environment
* OS Version: Ubuntu 20.04.5 running on Windows WSL2
* ROS Distro: Noetic running on Conda|Robostack
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
```
rviz version 1.14.5
compiled against Qt version 5.12.9
compiled against OGRE version 1.10.12 (Xalafu)
Forcing OpenGl version 0.
```

* If source build, which git commit? N/A
* System locale, i.e. the output of `echo "$LANG $LC_NUMERIC"`: en
* Before reporting a rendering issue, try running RViz with `LANG=C rviz`!: Tried that. Did not help

rhaschke, 2022-12-05:
rviz is looking for the urdfdom 1.1 version it was compiled for. If you are running a conda environment, you should install and run rviz from that environment too. You can't mix the default Debian environment with your conda environment.
If you have rviz installed in your conda env and you get this error, you should file a bug report to RoboStack.
Note, there is no need to run a conda env for ROS Noetic in WSL2. You can simply follow the [default ROS install instructions](http://wiki.ros.org/noetic/Installation/Ubuntu).

sspatank, 2022-12-05:
Thank you. I have posted my issue there. I needed to use conda because I need a different python version for another project on WSL2.  There has been no mixing of Debian and conda environments and so I think it's probably a robostack conda issue. I'll link to that issue once it gets resolved for any future travelers.

sspatank, 2022-12-05:
[This comment on my issue raised in Robostack/ros-noetic fixes this issue.](https://github.com/RoboStack/ros-noetic/issues/311#issuecomment-1338192691)

closed 2022-12-06 by rhaschke