ros-visualization/rviz #1776: Not save pointcloud map

alvaropascuaI, 2022-12-12:
I launch topics that i need for see in rviz pointcloud, but dont save point cloud map, what i need configure ?

I launch camera with this `roslaunch realsense2_camera rs_camera.launch camera:=cam_2 serial_no:=153222070827 filters:=spatial,temporal,pointcloud`

I launch rviz with `rviz` on terminal

[image]

### Your environment
* OS Version: Ubuntu 20.04
* ROS Distro: Noetic

alvaropascuaI, 2022-12-12:
What I want is for a map to be generated with the points that the rviz shows me, as with RTABMAP, but the problem is that a map is not generated from those point clouds

rhaschke, 2022-12-12:
I don't know about `RTABMAP`. In any case, rviz doesn't perform any processing, it just displays information available in your ROS system. So, if you need to process the point cloud into some map, this should be handled elsewhere.

alvaropascuaI, 2022-12-12:
I understand, thank you so much i will search more information

closed 2022-12-12 by rhaschke