ros-visualization/rviz #1777: rviz would not work on noetic and melodic because of the SIP4 bindings not working

natto1784, 2022-12-22:
This issue is a semi-copy of https://github.com/ros-visualization/python_qt_binding/issues/121
rviz1 would not work because it only works on SIP; and SIP4, which is what python-qt-binding seems to support for noetic and melodic (and apparently rolling and other recent distros as well), cannot work with recent libraries of pyqt5. This issue encourages the addition/reenabling of pyside/shiboken if possible, as that should(?) work with python-qt-binding. thanks.

### Your environment
* OS Version: e.g. NixOS 23.05.20221205.543070e (Stoat) x86_64
* ROS Distro: Noetic
* If source build, which git commit?
 6a234b4404c5ee71051f0b338ab8d5d2d931c27b
* System locale, i.e. the output of `echo "$LANG $LC_NUMERIC"`:
 en_US.UTF-8

rhaschke, 2023-01-03:
I'm not quite sure what exactly you ask for here. rviz' python bindings based on SIP are working correctly, aren't they?
If not, please provide a detailed error report.
If you are asking to re-add support for PySide/Shiboken, have a look at #1526. This PR got stuck due to issues in the code generated by shiboken. You are very welcome to check whether that has been fixed since then...

natto1784, 2023-01-03:
> I'm not quite sure what exactly you ask for here. rviz' python bindings based on SIP are working correctly, aren't they?

No, it appears they are not. PyQT5 dropped support for SIP4 and SIP4 is what python-qt-binding uses.

>If you are asking to re-add support for PySide/Shiboken

Yes that is the best course of action right now

rhaschke, 2023-01-03:
> > rviz' python bindings based on SIP are working correctly, aren't they?
> 
> No, it appears they are not. PyQT5 dropped support for SIP4 and SIP4 is what python-qt-binding uses.

Can you please provide a link documenting this change? As far as I can see, both `python-qt-binding` and rviz' python bindings compile as expected.

natto1784, 2023-01-03:
Hey, I looked more into it and I must say that I am sorry for prematurely opening this issue but the problem still remains nonetheless. There is incompatibility with the minimum_glibc versions with pyqt5 and sip4, so there can only be two fixes to this
1. reenabling shiboken/pyside2
2. waiting for upstream sip4 fix

The latter case is more of a sip4 problem than an rviz one, so if we are to only wait for the upstream fix, this issue is safe to be closed.
source: https://www.riverbankcomputing.com/pipermail/pyqt/2022-August/044845.html

rhaschke, 2023-01-03:
> There is incompatibility with the minimum_glibc versions with pyqt5 and sip4

Maybe that's only an issue in your `NixOS` distro? I didn't observe (or hear of) any real issues in Ubuntu 20.04.

rhaschke, 2023-01-03:
> reenabling shiboken/pyside2

This still fails due to a bug in the wrapper code generated by shiboken. So, that's no viable option yet.

closed 2023-01-03 by rhaschke

natto1784, 2023-01-03:
> > There is incompatibility with the minimum_glibc versions with pyqt5 and sip4
> 
> Maybe that's only an issue in your `NixOS` distro? I didn't observe (or hear of) any real issues in Ubuntu 20.04.

Nope, but it probably works on ubuntu 20.04 because of the predistributed binaries. Anyway, I guess waiting for an upstream SIP4 fix is better for now.

kjeremy, 2023-02-07:
@natto1784 is the SIP4 project aware?

natto1784, 2023-02-07:
> @natto1784 is the SIP4 project aware?

Yes, they are. However, as far as I know, they have decided to not work on maintaining SIP4 support for pyqt5, although I might be wrong about that. But even so, this should not be anything patching pyqt5 cannot fix, ([AUR](https://aur.archlinux.org/packages/python-pyqt5-sip4) already does this); though it would be better to avoid patching and enable PySide2 support here instead. What would be even better would be using rviz2 instead.