ros-visualization/rviz #1778: Parts of EnumProperty are hidden when the display width is less than the string length.

ysl-design, 2023-01-04:
Hi, I found that in the noetic version, when EnumProperty is selected, parts of it are hidden when the display width is less than the length of the string.
[image]
This problem does not occur in the melodic version, probably because the qt version is different. In the melodic version, like this:
[image]
I think adding cb->setSizeAdjustPolicy(QComboBox::AdjustToMinimumContentsLengthWithIcon) to the EnumProperty::createEditor function, the same settings as in EditableEnumProperty::createEditor should solve this problem, don't you think?
### Your environment
* OS Version: Ubuntu 20.04
* ROS Distro: Noetic
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
    ```
    [ INFO] [1672835556.680492000]: rviz version 1.14.19
    [ INFO] [1672835556.680552000]: compiled against Qt version 5.12.8
    [ INFO] [1672835556.680585900]: compiled against OGRE version 1.9.0 (Ghadamon)
    [ INFO] [1672835556.690115400]: Forcing OpenGl version 0.
    [ INFO] [1672835556.900242300]: Stereo is NOT SUPPORTED
    [ INFO] [1672835556.900664600]: OpenGL device: llvmpipe (LLVM 12.0.0, 256 bits)
    [ INFO] [1672835556.900827300]: OpenGl version: 3.1 (GLSL 1.4).
    ```
* If source build, which git commit?
* System locale, i.e. the output of `echo "$LANG $LC_NUMERIC"`: zh_CN.UTF-8

rhaschke, 2023-01-04:
You are right, non-editable EnumProperties don't require the dropdown icon to be visible as you can click everywhere in the combobox to open the dropdown. Enforcing to show the icon reduces readability in my opinion:

[image]
vs.
[image]

ysl-design, 2023-01-05:
Yes, this will show more information. Okay, I'll close the issue

closed 2023-01-05 by ysl-design