ros-visualization/rviz #1779: How to enable GPU for rviz

ZhangZh3ng, 2023-01-20:
I need to render a large number of point cloud and marker array in rviz, now the rviz fps is very low(< 1fps),  I tried to open a terminal with GPU(mine is Nvidia1050Ti) and then run rviz in that terminal, but it doesn't help. I found the the GPU memory usage by rviz is very low(4 Mb) which is abnormal. So I want to know is there any way to enable GPU for rviz?

### Your environment
* OS Version:  Ubuntu 20.04
* ROS Distro: [Noetic]
* RViz
```
$ rviz
[ INFO] [1674207641.917945501]: rviz version 1.14.19
[ INFO] [1674207641.918000845]: compiled against Qt version 5.12.8
[ INFO] [1674207641.918017279]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1674207641.929910597]: Forcing OpenGl version 0.
[ INFO] [1674207642.369170630]: Stereo is NOT SUPPORTED
[ INFO] [1674207642.369286811]: OpenGL device: NVIDIA GeForce GTX 1050 Ti/PCIe/SSE2
[ INFO] [1674207642.369384268]: OpenGl version: 4.6 (GLSL 4.6).
QFileSystemWatcher::removePaths: list is empty
QFileSystemWatcher::removePaths: list is empty
```

rhaschke, 2023-01-20:
The output `OpenGL device: NVIDIA GeForce GTX 1050 Ti/PCIe/SSE2` already indicates that the GPU is used for rendering.
Maybe you GPU is already overloaded with (pre)processing of the point-cloud data, i.e. receiving and preparing data for display? This is done in a separate thread for each PC display.

ZhangZh3ng, 2023-01-20:
> The output `OpenGL device: NVIDIA GeForce GTX 1050 Ti/PCIe/SSE2` already indicates that the GPU is used for rendering. Maybe you GPU is already overloaded with (pre)processing of the point-cloud [...]

Thank you for your reply, maybe I need to decrease to point number.

rhaschke, 2023-01-21:
Yes.

closed 2023-01-21 by rhaschke