ros-visualization/rviz #1780: Display Panel in ROS Noetic Docker Container Rendering Issue

ijnek, 2023-02-08:
When dragging display panels around in rviz in a docker container, the rendering acts up strange.

file:///home/ijnek/2023-02-07%2022-17-21.mp4

I **have not** been able to reproduce this on:
* a locally installed ROS Noetic installation
* ros-melodic-full docker container

but **have** managed to reproduce it using the same docker container across two machines with different gpus (integrated and nvidia).

```sh
docker run --rm -it --privileged --net=host --ipc=host \
--device=/dev/dri:/dev/dri \
-v /tmp/.X11-unix:/tmp/.X11-unix -e DISPLAY=$DISPLAY \
-v $HOME/.Xauthority:/home/$(id -un)/.Xauthority -e XAUTHORITY=/home/$(id -un)/.Xauthority \
-e ROS_IP=127.0.0.1 \
osrf/ros:noetic-desktop
```

### Your environment
* OS Version: Ubuntu 22.04 (running ros-noetic-desktop docker image)
* ROS Distro: Noetic
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
    ```
    [ INFO] [1675829167.064978650]: rviz version 1.14.19
    [ INFO] [1675829167.065029626]: compiled against Qt version 5.12.8
    [ INFO] [1675829167.065041080]: compiled against OGRE version 1.9.0 (Ghadamon)
    [ INFO] [1675829167.075989223]: Forcing OpenGl version 0.
    [ INFO] [1675829167.159583133]: Stereo is NOT SUPPORTED
    [ INFO] [1675829167.159629826]: OpenGL device: Mesa Intel(R) Graphics (ADL GT2)
    [ INFO] [1675829167.159641504]: OpenGl version: 4.6 (GLSL 4.6) limited to GLSL 1.4 on Mesa system.
    ```

rhaschke, 2023-02-08:
Looks more like a Qt than an rviz issue. How do other Qt apps behave that have similar panels?

ijnek, 2023-02-08:
Thanks for the quick response @rhaschke, I haven't been able to reproduce this with rqt when I drag panels around.

l1va, 2023-03-24:
Hey, I think I found the workaround. 
The problem in the qt. And here is the bug https://bugreports.qt.io/browse/QTBUG-81723 
I upgraded the qt to 5.12.10 with the next steps: 
   ```
       sudo add-apt-repository ppa:beineri/opt-qt-5.12.10-focal
       sudo apt-get update
       sudo apt-get install qt512charts-no-lgpl qt512svg qt512xmlpatterns qt512tools qt512translations    qt512graphicaleffects qt512quickcontrols2 qt512wayland qt512websockets qt512serialbus qt512serialport qt512location qt512imageformats qt512script qt512scxml qt512gamepad qt5123d 
       source /opt/qt512/bin/qt512-env.sh
```

Then cloned rviz to catkin_ws/src and built with 'catkin build rviz'.    
After that rviz is working.

l1va, 2023-03-27:
one more thing. 
We found that it is not needed to rebuild the rviz. Only install and source qt.

ijnek, 2023-03-30:
Hi @l1va, thanks for figuring out this issue! I tried your workaround and things seem to be working :+1:

closed 2023-03-30 by ijnek