ros-visualization/rviz #1781: no matching function for call to ‘aiMatrix4x4t<float>::Decompose(aiVector3D&, aiVector3D&, float&, aiVector3D&)

andre-f-moura, 2023-02-15:
Hello, 
I'm trying to build ROS Noetic from source on Fedora 34, following the instructions on the [ROS Wiki](http://wiki.ros.org/Installation/Source).
I'm encountering the following error while building rviz. I also tried to download directly from the rviz git source, but observed the same error.

### My environment
* OS Version: Fedora 34 (Workstation Edition)
* ROS Distro: Noetic (from source)
    ```
 ==> Processing catkin package: 'rviz'
==> Building with env: '/opt/ros/noetic/env.sh'
Makefile exists, skipping explicit cmake invocation...
==> make cmake_check_build_system in '/home/andre/ros_catkin_ws/build_isolated/rviz'
==> make -j8 -l8 in '/home/andre/ros_catkin_ws/build_isolated/rviz'
[  0%] Built target std_msgs_generate_messages_cpp
[  0%] Built target std_msgs_generate_messages_nodejs
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target std_msgs_generate_messages_lisp
[  0%] Built target std_msgs_generate_messages_eus
[  0%] Built target tf_generate_messages_nodejs
[  0%] Built target std_srvs_generate_messages_nodejs
[  0%] Built target std_srvs_generate_messages_lisp
[  0%] Built target rosgraph_msgs_generate_messages_nodejs
[  0%] Built target std_srvs_generate_messages_eus
[  0%] Built target actionlib_generate_messages_cpp
[  0%] Built target nav_msgs_generate_messages_eus
[  0%] Built target nav_msgs_generate_messages_lisp
[  0%] Built target actionlib_msgs_generate_messages_py
[  0%] Built target std_srvs_generate_messages_py
[  0%] Built target actionlib_msgs_generate_messages_eus
[  0%] Built target roscpp_generate_messages_cpp
[  0%] Built target map_msgs_generate_messages_nodejs
[  0%] Built target rosgraph_msgs_generate_messages_eus
[  0%] Built target _rviz_generate_messages_check_deps_SendFilePath
[  0%] Built target roscpp_generate_messages_py
[  0%] Built target roscpp_generate_messages_eus
[  0%] Built target sensor_msgs_generate_messages_cpp
[  0%] Built target geometry_msgs_generate_messages_eus
[  0%] Built target nav_msgs_generate_messages_nodejs
[  0%] Built target sensor_msgs_generate_messages_nodejs
[  0%] Built target tf2_msgs_generate_messages_eus
[  0%] Built target geometry_msgs_generate_messages_py
[  0%] Built target sensor_msgs_generate_messages_py
[  0%] Built target map_msgs_generate_messages_py
[  0%] Built target actionlib_generate_messages_lisp
[  0%] Built target roscpp_generate_messages_lisp
[  0%] Built target rosgraph_msgs_generate_messages_py
[  0%] Built target geometry_msgs_generate_messages_cpp
[  0%] Built target tf2_msgs_generate_messages_py
[  0%] Built target nav_msgs_generate_messages_cpp
[  0%] Built target visualization_msgs_generate_messages_eus
[  0%] Built target geometry_msgs_generate_messages_nodejs
[  0%] Built target visualization_msgs_generate_messages_lisp
[  0%] Built target tf2_msgs_generate_messages_cpp
[  0%] Built target rosgraph_msgs_generate_messages_lisp
[  0%] Built target tf_generate_messages_eus
[  0%] Built target roscpp_generate_messages_nodejs
[  0%] Built target geometry_msgs_generate_messages_lisp
[  0%] Built target tf_generate_messages_cpp
[  0%] Built target map_msgs_generate_messages_cpp
[  0%] Built target map_msgs_generate_messages_lisp
[  0%] Built target actionlib_msgs_generate_messages_cpp
[  0%] Built target actionlib_generate_messages_eus
[  0%] Built target rosgraph_msgs_generate_messages_cpp
[  0%] Built target std_srvs_generate_messages_cpp
[  0%] Built target sensor_msgs_generate_messages_eus
[  0%] Built target sensor_msgs_generate_messages_lisp
[  0%] Built target visualization_msgs_generate_messages_cpp
[  0%] Built target map_msgs_generate_messages_eus
[  0%] Built target actionlib_generate_messages_nodejs
[  0%] Built target actionlib_msgs_generate_messages_nodejs
[... 57 lines cut ...]
  238 |     transform.Decompose(scaling, axis, angle, pos);
      |     ~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/assimp/types.h:514,
                 from /usr/include/assimp/Importer.hpp:53,
                 from /home/andre/ros_catkin_ws/src/rviz/src/rviz/mesh_loader.cpp:58:
/usr/include/assimp/matrix4x4.inl:302:13: note: candidate: ‘void aiMatrix4x4t<TReal>::Decompose(aiVector3t<TReal>&, aiQuaterniont<TReal>&, aiVector3t<TReal>&) const [with TReal = float]’
  302 | inline void aiMatrix4x4t<TReal>::Decompose (aiVector3t<TReal>& scaling, aiQuaterniont<TReal>& rotation,
      |             ^~~~~~~~~~~~~~~~~~~
/usr/include/assimp/matrix4x4.inl:302:13: note:   candidate expects 3 arguments, 4 provided
[ 11%] Building CXX object src/rviz/CMakeFiles/rviz.dir/ogre_helpers/ogre_render_queue_clearer.cpp.o
make[2]: *** [src/rviz/CMakeFiles/rviz.dir/build.make:342: src/rviz/CMakeFiles/rviz.dir/mesh_loader.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:2612: src/rviz/CMakeFiles/rviz.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
<== Failed to process package 'rviz': 
  Command '['/opt/ros/noetic/env.sh', 'make', '-j8', '-l8']' returned non-zero exit status 2.

Reproduce this error by running:
==> cd /home/andre/ros_catkin_ws/build_isolated/rviz && /opt/ros/noetic/env.sh make -j8 -l8

Command failed, exiting.

    ```
* Git commit: d122ab1
* System locale : en_US.UTF-8

rhaschke, 2023-02-15:
You are using an incompatible version of `libassimp`.

andre-f-moura, 2023-02-15:
Any suggestion on how to find and install a compatible version? 
Thank you for your reply.

rhaschke, 2023-02-15:
Have a look at the [_targeted_ platforms](https://www.ros.org/reps/rep-0003.html#noetic-ninjemys-may-2020-may-2025).
There, Fedora 32 is listed for Noetic. Use a corresponding assimp from that version.

closed 2023-02-15 by rhaschke