ros-visualization/rviz #1782: RVIZ render point efficently question

peter5232, 2023-03-07:
Describe your issue here and explain how to reproduce it.
I want to ask a question. I use glVertex3f when I use the QOpenGLWidget to call the OpenGL command to render the point cloud. But when I render nearly 100 million points, the whole program gets stuck. What measures does Rviz use to render the point cloud, and has it been thinned?
### Your environment
* OS Version: e.g. Ubuntu 18.04
* ROS Distro: [Kinetic | Melodic | Noetic]
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
    ```
    paste your console output here
    ```
* If source build, which git commit?
* System locale, i.e. the output of `echo "$LANG $LC_NUMERIC"`:
  Before reporting a rendering issue, try running RViz with `LANG=C rviz`!

rhaschke, 2023-03-07:
> I use glVertex3f when I use the QOpenGLWidget to call the OpenGL command to render the point cloud.

Are you talking about an rviz-independent program to render a point cloud?

> But when I render nearly 100 million points, the whole program gets stuck.

Which program? rviz? Your own?

> What measures does Rviz use to render the point cloud, and has it been thinned?

Are you asking for advice to improve your own program or are you complaining about poor performance of rviz?
There were some recent attempts to improve rviz rendering performance. See #1725 and #1733. Unfortunately they were not finished, mainly due to issues with old Ogre, which is used for rendering under the hood.
So, if you want advice for your own program: Have a look into Ogre. rviz uses vertex shaders to render point clouds.

rhaschke, 2023-03-29:
Closing due to inactivity.

closed 2023-03-29 by rhaschke