ros-visualization/rviz #1784: rviz::YamlConfigReader relative path

GavinGaoYan, 2023-03-10:
rviz::YamlConfigReader reader;
reader.readFile(demo,"/xxx/xxx/xx"); work

but

reader.readFile(demo,"../xx");    dead

what is the reason

GavinGaoYan, 2023-03-10:
[00049](http://docs.ros.org/en/hydro/api/rviz/html/c++/classrviz_1_1YamlConfigReader.html#a92ef5d06ca0a80a22e3e03b69319ace0) void [YamlConfigReader::readFile](http://docs.ros.org/en/hydro/api/rviz/html/c++/classrviz_1_1YamlConfigReader.html#a92ef5d06ca0a80a22e3e03b69319ace0)( [Config](http://docs.ros.org/en/hydro/api/rviz/html/c++/classrviz_1_1Config.html)& config, const QString& filename )
00050 {
00051   std::ifstream in( qPrintable( filename ));
00052   [readStream](http://docs.ros.org/en/hydro/api/rviz/html/c++/classrviz_1_1YamlConfigReader.html#aa1ca3a0eeafcb69f4b2a8f9e8139affc)( config, in, filename );
00053 }

rhaschke, 2023-03-10:
Probably the _current working_ directory is not pointing to where you expect it to point. There is no fundamental reason that this could fail.

closed 2023-03-10 by rhaschke