ros-visualization/rviz #1785: Suggestion: Arrow List Marker

rr-tom-noble, 2023-03-16:
Hi,

I've found myself frequently using arrow markers to visualise paths in rviz, however, using separate markers for each arrow feels messy and I imagine quite inefficient compared to using marker types which support lists of geometries. Furthermore, they tend to behave unreliably when rendering in large amounts i.e. >100 (usually frame transformation issues despite the frames existing).

I've tried switching over to the line strip geometry type, however, I've found it lacking in a couple of ways for my use case:

- Using arrows gives a better sense of direction, which is nice from a usability point of view.
- The "flat" appearance of the line tends to cause the visual for the path and the mesh that the path is being planned over to interfere with one another. I've found that the 3d arrows tend to look better and produce more reliable results for this.

With this in mind, I'm wondering if it'd be useful to add an arrow list marker type? I'm imagining that properties such as scaling and colour would apply to all arrows in the list so that they're uniformly shaped, and the points field would contain N points, defining N-1 arrows tip to tail.

I'm happy to attempt raising a PR if this sounds like a sensible idea

rhaschke, 2023-03-17:
I understand your need. As far as I understand it would be a variant of `LINE_STRIP`: instead of simply drawing a line, draw the arrow strip. Such a PR is definitely welcome. However, it's quite different from the existing `[SPHERE|CUBE|POINT]_LIST`, whose geoms all share the same orientation. Thus, I suggest naming the new type `ARROW_STRIP`.

rr-tom-noble, 2023-03-20:
Thanks for the advice. I'll try to get a PR raised some time this week. Could you let me know if there are any automated tests / additional documentation I should add before opening a PR?

rr-tom-noble, 2023-03-21:
@rhaschke I've implemented an `ArrowStripMarker` class which implements the `onNewMessage()` method. Could you let me know which other files would need updating to hook everything up to use the new class?

Thanks!

rhaschke, 2023-03-21:
I think you just need to adapt [`createMarker()` in `marker_utils.cpp`](https://github.com/ros-visualization/rviz/blob/d122ab111213c3ad4ac50756e4fd933d01982263/src/rviz/default_plugin/marker_utils.cpp#L53) as well.