ros-visualization/rviz #1787: RobotModelDisplay: wrong orientation of rotation axis for some robots/joints

simonschmeisser, 2023-03-28:
[image]

For some robot models (UR, Techman, Doosan) the display of rotation axis is incorrect. The screenshot shows joint 6 or `wrist_3_joint` and should obviously point up or down, ie rotate around the end effector

Describe your issue here and explain how to reproduce it.

### Your environment
* OS Version: Ubuntu 22.04
* ROS Distro: One
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
    ```
[INFO] [1679993270.044010752]: rviz version 1.14.19
[INFO] [1679993270.044065063]: compiled against Qt version 5.15.7
[INFO] [1679993270.044078448]: compiled against OGRE version 1.9.0 (Ghadamon)
[INFO] [1679993270.054538631]: Forcing OpenGl version 0.
[INFO] [1679993270.269967668]: Stereo is NOT SUPPORTED
[INFO] [1679993270.270020407]: OpenGL device: RENOIR (renoir, LLVM 15.0.6, DRM 3.47, 5.19.0-35-generic)
[INFO] [1679993270.270035786]: OpenGl version: 4,6 (GLSL 4,6) limited to GLSL 1.4 on Mesa system.

    ```
* If source build, which git commit?
* System locale, i.e. the output of `echo "$LANG $LC_NUMERIC"`:
  Before reporting a rendering issue, try running RViz with `LANG=C rviz`!

rhaschke, 2023-03-28:
I ran into this issue recently as well and pushed a fixup: 88dd507a299b9c40ea494010810d18db16ea6366
Can you please verify if that fixes your issue. If so, I should maybe trigger a new release...

rhaschke, 2023-03-28:
Looks like https://github.com/ros-visualization/rviz/commit/88dd507a299b9c40ea494010810d18db16ea6366 addressed another issue. Please review #1788, which should fix this one here.

simonschmeisser, 2023-03-29:
closed via #1788

PR #1788 (2023-03-29): Fix orientation of joint axis arrow
Fixes #1787
closed 2023-03-29 by simonschmeisser